ros2/rclc #11: Status of rclc?

colinxs, 2019-03-01:
Hello! 

I'm really looking forward to ROS2's C API and was wondering what the current status of the project is? What's the best way to stay up to date in the future?
Thanks!

tfoote, 2019-03-01:
rclc is not under active development at the moment. We are focused on rclcpp and rclpy as the primary client libraries. In the future we hope to be able to return to this and fill it out. As well as refactor as much supporting code out of the c++ implementation into the rcl libraries.

If someone is interested in picking it up that would be great. But so far there hasn't been significant support or demand for this client library.

mjcarroll, 2019-03-14:
@colinxs I believe that @tfoote has sufficiently answered your question, so I'm going to close this out.  If you have any additional questions, please feel free to ask in the relevant location based on our [support guidelines](http://wiki.ros.org/Support)

closed 2019-03-14 by mjcarroll