ros2/rclc #15: Support for clients and services in the rclc_executor

ralph-lange, 2020-03-16:
Add support for clients and services to the rclc_executor, in the style of rclc_executor_add_subscription (cf. https://github.com/ros2/rclc/blob/master/rclc/include/rclc/executor.h#L208).

jayhou, 2020-07-02:
> Add support for clients and services to the rclc_executor, in the style of rclc_executor_add_subscription (cf. https://github.com/ros2/rclc/blob/master/rclc/include/rclc/executor.h#L208).

Is there any plan to support client and service in rclc_executor?

ralph-lange, 2020-07-02:
Yes. My colleague @JanStaschulat is already working on this in the branch https://github.com/micro-ROS/rclc/tree/feature/support_for_service_clients.

jayhou, 2020-07-03:
> Yes. My colleague @JanStaschulat is already working on this in the branch https://github.com/micro-ROS/rclc/tree/feature/support_for_service_clients.

thanks, so great!

JanStaschulat, 2021-01-20:
Services, clients and guard conditions were added in PR https://github.com/ros2/rclc/pull/26

closed 2021-01-20 by JanStaschulat