ros2/rclc #31: Update access permissions

JanStaschulat, 2021-01-20:
@wjwwood 
Hi William, there was a change at eProsima. Julian (user name: julibert) left the company and Pablo (user name: pablogs9) is the new maintainer instead. Could you please remove Julian from the rclc-repo permissions and add Pablo instead? This change was documented in  https://github.com/ros2/rclc/blob/master/rclc/package.xml#L8

I am not able to change these access permissions. 
Thanks, Jan.

clalancette, 2021-01-20:
@JanStaschulat You should have permissions to remove @julibert from the team by clicking on the link here: https://github.com/orgs/ros2/teams/rclc/members .  Is that not the case?

Additionally, Pablo is already marked as an Admin for rclc.  So he should already have access.  That being said, probably the best thing to do is to invite Pablo to be a member of the ros2 organization, and then we can add him to the rclc team.  I'm going to go ahead and do the invitation; once he accepts, you should have the permissions to add him to the rclc team.

JanStaschulat, 2021-01-20:
@clalancette No, that is not the case. I don't have the permission to remove anyone. In the section "Manage access"  I can only add someone. I don't see any remove button. Same is true for the rclc-team. There I can only add people but not remove. 

Yes, I just added him to the rclc-team, there he is now only listed as member, but not as maintainer.

clalancette, 2021-01-20:
I really don't know why; you have all of the permissions that I can think to give you.

Regardless, I've just made Pablo a Maintainer on the rclc team and removed Julian.  Pablo should now have the same permissions as you do.  I'm going to close this out, but feel free to reopen if there is still something else to do here.

closed 2021-01-20 by clalancette

JanStaschulat, 2021-01-20:
Thank you. Permissions are correct now.