ros2/rclc #33: RCLCCHECK

norro, 2021-01-21:
Suggestion:
Make the current `RCCHECK` macros (see [here](https://github.com/ros2/rclc/blob/6e45370f4bb5434e5bd81974829795284604cf4b/rclc_examples/src/example_client_node.c#L24-LL27) for example) public as `RCCHECK`/`RCLCCHECK`/`RCLC_CHECK` macro.

Rationale:
I end up introducing the same code again and again in almost all of my rclc files as well.

norro, 2021-01-21:
@JanStaschulat what do you think?

JanStaschulat, 2021-01-21:
Sure, you can add the macro in rclc/include/rclc/types.h like this [macro](https://github.com/micro-ROS/rclc/blob/b8ef4a6c96f3c80f1fbfd23e8127e3948d44d011/rclc/include/rclc/types.h#L42).

JanStaschulat, 2021-01-22:
Please create PR to micro-ros/rclc.

JanStaschulat, 2021-02-08:
Add also the error message in RCCHECK RCSOFTCHECK, like [here](https://github.com/micro-ROS/nuttx_apps/blob/8f2e0f6bfccc0d249deb730a36f17da1cb49b006/examples/uros_rbs/main_rbs.c#L33)

norro, 2021-02-17:
Closing as per discussion here: https://github.com/micro-ROS/rclc/pull/66

closed 2021-02-17 by norro