ros2/rclc #35: Allow passing context to service callbacks

norro, 2021-01-22:
Currently, service callbacks get
1. either exactly two arguments: pointer to the request and pointer to the response
1. or three arguments: pointer to the request, pointer to the response, and the request id

This is rather limiting, as it does not allow to provide any additional context to the callback. E.g., in case of the lifecycle node: when the state change service is called, I currently don't have the possibility to get the lifecycle node on which the state change was called.

I propose to add a third option, in addition to the two introduced above, where a third `void *` argument allows passing additional context, e.g., the lifecycle node in the example above.

JanStaschulat, 2021-01-25:
As discussed, I agree.  Thanks for suggesting this enhancement for service callbacks.

BrettRD, 2021-01-28:
This need also applies to subscriptions
I have an application where I'm opening and closing multiple pipelines that each start with a ROS subscription.
Multiple callbacks and global variables aren't an option.
The workaround to shim a context pointer in is pretty awful

JanStaschulat, 2021-01-28:
@BrettRD Sure,  subscriptions can be extended the same way.

norro, 2021-02-17:
Resolved with https://github.com/ros2/rclc/pull/45

closed 2021-02-17 by norro

JanStaschulat, 2021-02-17:
@BrettRD please create an issue at https://github.com/ros2/rclc/issues to request subscriptions with context for the rclc-executor.