ros2/rclc #40: Service call is processed twice

norro, 2021-01-27:
In an rclc application of mine, but also in the example service node (when applying [this fix](https://github.com/ros2/rclc/issues/39#issuecomment-768215233)), I see that the service is processed twice. I do not have the appropriate rcl(c) knowledge / debugging abilities to understand the source of this issue.

**Steps to reproduce:**
1. Apply [this fix](https://github.com/ros2/rclc/issues/39#issuecomment-768215233) to the example service node 
1. Execute the example service node 
1. Call the service once, e.g., with `ros2 service call /addtwoints example_interfaces/srv/AddTwoInts "{a: 17, b: 42}"`

**Expected output** of the example service node:
```
Service request value: 17 + 42
```

**Observed output** of the example service node:
```
Service request value: 17 + 42
Service request value: 17 + 42
```

norro, 2021-02-17:
Can no longer reproduce with current rclc master.

closed 2021-02-17 by norro

karl-schulz, 2022-09-27:
For anyone seeing this issue: I had a similar behavior when I accidentally attached two identical services to a node. RCL doesn't warn about this and just calls the service callback twice.