ros2/rclc #49: Fix unit test on build farm for new service functions

JanStaschulat, 2021-02-16:
@norro 
Hi Arne, could you have a look, why these unit tests with services fail?
https://build.ros2.org/job/Rpr__rclc__ubuntu_focal_amd64/71/#showFailuresLink

JanStaschulat, 2021-02-16:
@norro: The expected sequence id is 1, however for Rolling release in the docker container the returned sequence id is different in each unit test. As this is not  part of the executor functionality, but rather a RCL/RMW issue, I would propose to omit this ASSERT in all relevant unit_tests with services/clients:

https://github.com/ros2/rclc/blob/c5b0d3d08a6f4fa70d3e056a4f3ff70158c808c3/rclc/test/rclc/test_executor.cpp#L2081

norro, 2021-02-17:
Agreed, I wouldn't have a clue how to resolve this.
Should we / can we report an issue upstream (in rcl/rmw) to have this fixed upstream?

JanStaschulat, 2021-02-17:
Yes, that is a good idea. you could also point out, that it is only an issue on the build farm and not in a normal `colcon test` with Github Action.

norro, 2021-02-19:
@JanStaschulat Could you please write the upstream issue in rcl/rmw? I don't have enough knowledge of rcl/rmw to be able to phrase the issue, I'm afraid.
After the issue is created, I suggest to close this issue.

JanStaschulat, 2021-02-22:
@norro created an issue for ros2/rclc https://github.com/ros2/rcl/issues/892. I would keep the issue open - because I referenced it in the ros2/rcl issue.

JanStaschulat, 2021-02-22:
@norro. As explained in the issue https://github.com/ros2/rcl/issues/892, the specification of `rcl_send_request` does not say to which value `seq_id` is initialized to. It is only guaranteed to be unique. 

Therefore we can not check this value (even though it is always "1" for Dashing and Foxy. So we can keep the unit test the way the are. I will remove these commented out lines. Then we can close this issue.

closed 2021-02-23 by JanStaschulat