ros2/rclc #50: Allow passing context to subscription callbacks

BrettRD, 2021-02-17:
Following from https://github.com/ros2/rclc/issues/35 add context pointers to service calls, I'd like to use subscriptions with context pointers too

The change would re-use (and re-name) the `void * service_context;` added in https://github.com/ros2/rclc/pull/45, and add the following:

```
typedef void (* rclc_subscription_callback_with_context_t)(const void *, void *);
```

```
rcl_ret_t
rclc_executor_add_subscription_with_context(
  rclc_executor_t * executor,
  rcl_subscription_t * subscription,
  void * msg,
  rclc_callback_t callback,
  void * context,
  rclc_executor_handle_invocation_t invocation);
```

I have most of a patch, I'll re-base it onto https://github.com/ros2/rclc/pull/45 in the next couple of weeks and submit a pull to micro-ros

JanStaschulat, 2021-02-26:
Just for information: in future create a feature branch from ros2/rclc and create a pull request directly to ros2/rclc. 
See also ticket https://github.com/micro-ROS/rclc/issues/75

BrettRD, 2021-06-30:
implemented in https://github.com/ros2/rclc/pull/107, but needs https://github.com/ros2/rclc/pull/106 first
sorry about the pull to foxy, that's the only branch I'm set up to test on.

JanStaschulat, 2021-06-30:
@BrettRD, just let me know, when you are done, then I'll do a code review.

BrettRD, 2021-06-30:
I think it's ready, I was expecting way more merge conflicts

closed 2021-07-17 by JanStaschulat

pedropiacenza, 2024-11-10:
How would I go about implementing this? I'm working on a microros application to control dynamixel servos and want to organize the code running on my Teensy 4.1 as a class. 
In such scenario, the rcl node, allocator, executor, subscriptions and publishers are all member variables of the main class. 

During the initialization method of the class, I am initializing the subscriber like so:

```
// Create subscriber for "servo_command"
rclc_subscription_init_default(
    &servo_cmd_subscriber_,
    &node_,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, UInt16MultiArray),
    "rh3_servo_pos_commands");
```
This is how I try to add the subscription to the executor using rclc_executor_add_subscription_with_context:

```
// Init executor
executor_ = rclc_executor_get_zero_initialized_executor();
rclc_executor_init(&executor_, &support_.context, 3, &allocator_);
rclc_executor_add_subscription_with_context(
    &executor_, 
    &servo_cmd_subscriber_, 
    &new_cmd_msg_, 
    &commandsCallback, 
    this,
    ON_NEW_DATA);
```

Here commandsCallback is a method of the main class, with the following header and implementation:

`void commandsCallback(const void * msgin);`

```
void RH3Controller::commandsCallback(const void * msgin){
  const std_msgs__msg__UInt16MultiArray * msg = (const std_msgs__msg__UInt16MultiArray *)msgin; 
  new_cmd_msg_.layout = msg->layout;
  new_cmd_msg_.data = msg->data;
  printNewCommand();
  service_servos_ = true;
}
```

Is it even possible to have the callback point to a method inside the class?? How do I do this? 
This is the current error I get with the code shown above:

```
In file included from /home/pedro/code/Arduino/libraries/RH3_Controller/RH3_Controller.cpp:1:
/home/pedro/code/Arduino/libraries/RH3_Controller/RH3_Controller.cpp: In member function 'bool RH3Controller::initialize()':
/home/pedro/code/Arduino/libraries/RH3_Controller/RH3_Controller.cpp:107:6: warning: ISO C++ forbids taking the address of an unqualified or parenthesized non-static member function to form a pointer to member function.  Say '&RH3Controller::commandsCallback' [-fpermissive]
  107 |     &commandsCallback,
```
By the way, using &RH3Controller::commandsCallback as a parameter for rclc_executor_add_subscription_with_context instead of &commandsCallback also doesn't work.

BrettRD, 2024-11-11:
https://github.com/BrettRD/rclc/blob/d75c2f0e25798112049d22a72e907cf4e1be1039/rclc_examples/src/example_sub_context.c#L52
You've passed the `this` pointer as context which is a good start, your subscription callback needs to cast the context void pointer back into the same type as `this` before you can call class methods on it with `this->method()`.