ros2/rclc #65: Build error on Rolling

anaelle-sw, 2021-03-25:
- Hardware description: Ubuntu 20.04, ROS2 Rolling
- Installation type: micro-ROS Arduino installed from sources
- Version or commit hash: micro-ROS Arduino from Rolling branch (main)

#### Steps to reproduce the issue
Hello, 
I tried to build the latest version from micro-ROS Arduino, but a build error occurred:
```c
cd ~/Arduino/libraries
git clone -b main https://github.com/micro-ROS/micro_ros_arduino.git
cd ~/Arduino/libraries/micro_ros_arduino
sudo docker pull microros/micro_ros_arduino_builder:rolling
sudo docker run -it --rm -v $(pwd):/arduino_project microros/micro_ros_arduino_builder:rolling -p teensy32
```
This last command failed:
```c
Starting >>> rclc_lifecycle
--- stderr: rclc_lifecycle                                                                                                                                                                           
CMake Warning at /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /uros_ws/firmware/mcu_ws/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_runtime_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/rosidl_runtime_c/cmake/rosidl_runtime_cConfig.cmake:41 (include)
  /uros_ws/firmware/mcu_ws/install/share/lifecycle_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /uros_ws/firmware/mcu_ws/install/share/lifecycle_msgs/cmake/lifecycle_msgsConfig.cmake:41 (include)
  CMakeLists.txt:11 (find_package)

/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c: In function 'rclc_make_node_a_lifecycle_node':
/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:50:5: warning: passing argument 9 of 'rcl_lifecycle_state_machine_init' makes pointer from integer without a cast [-Wint-conversion]
     true,
     ^
In file included from /uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/include/rclc_lifecycle/rclc_lifecycle.h:24:0,
                 from /uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:17:
/uros_ws/firmware/mcu_ws/install/include/rcl_lifecycle/rcl_lifecycle.h:244:1: note: expected 'const rcl_lifecycle_state_machine_options_t * {aka const struct rcl_lifecycle_state_machine_options_t *}' but argument is of type 'int'
 rcl_lifecycle_state_machine_init(
 ^
/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:41:23: error: too many arguments to function 'rcl_lifecycle_state_machine_init'
   rcl_ret_t rcl_ret = rcl_lifecycle_state_machine_init(
                       ^
In file included from /uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/include/rclc_lifecycle/rclc_lifecycle.h:24:0,
                 from /uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:17:
/uros_ws/firmware/mcu_ws/install/include/rcl_lifecycle/rcl_lifecycle.h:244:1: note: declared here
 rcl_lifecycle_state_machine_init(
 ^
/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c: In function 'rcl_lifecycle_node_fini':
/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:233:13: error: too many arguments to function 'rcl_lifecycle_state_machine_fini'
   rcl_ret = rcl_lifecycle_state_machine_fini(
             ^
In file included from /uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/include/rclc_lifecycle/rclc_lifecycle.h:24:0,
                 from /uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:17:
/uros_ws/firmware/mcu_ws/install/include/rcl_lifecycle/rcl_lifecycle.h:278:1: note: declared here
 rcl_lifecycle_state_machine_fini(
 ^
make[2]: *** [CMakeFiles/rclc_lifecycle.dir/build.make:66: CMakeFiles/rclc_lifecycle.dir/src/rclc_lifecycle/rclc_lifecycle.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:108: CMakeFiles/rclc_lifecycle.dir/all] Error 2
make: *** [Makefile:144: all] Error 2
---
Failed   <<< rclc_lifecycle [1.12s, exited with code 2]
```

#### Additional information
@pablogs9 gave me the following information:
this is an RCLC problem because they have not updated their RCL API usage in Rolling. These are the changes:  [ros2/rcl@e9b588d](https://github.com/ros2/rcl/commit/e9b588d1ed4ea11bd667c25fb35e231883cadc90)

pablogs9, 2021-03-25:
Hello, @ralph-lange @JanStaschulat @norro could you please check that? I guess that there are changes in RCL lifecycle and incompatibility with RCLC Rolling.

Thanks a lot!

ralph-lange, 2021-03-25:
Yep, also fails on the build farm since yesterday: https://build.ros2.org/job/Rbin_uF64__rclc_lifecycle__ubuntu_focal_amd64__binary/37/ to https://build.ros2.org/job/Rbin_uF64__rclc_lifecycle__ubuntu_focal_amd64__binary/40/

pablogs9, 2021-03-25:
All micro-ROS CI are failing in Rolling. 

Does your team have some time to fix this @ralph-lange? Or maybe I should find a temporal solution like disabling this package for micro-ROS?

Sorry for the inconvenience.

norro, 2021-03-25:
@pablogs9 I probably can find some time today or tomorrow to fix rclc_lifecycle. /cc @ralph-lange

ralph-lange, 2021-03-25:
Thank you for taking care of this, @norro. The relevant change to rcl_lifecycle can be found at https://github.com/ros2/rcl/pull/882.

pablogs9, 2021-03-25:
Thanks a lot @norro @ralph-lange !!

JanStaschulat, 2021-03-26:
documentation: 
source code changes in rclc_lifecycle: https://github.com/ros2/rclc/pull/66
updated CI-build job for Rolling: https://github.com/ros2/rclc/pull/66
updated CI-build job for Dashing, Foxy https://github.com/ros2/rclc/pull/67/files
updated ROS2 build farm bloom-release for Dashing (to use foxy branch): https://github.com/ros/rosdistro/pull/28855

JanStaschulat, 2021-03-26:
@anaelle-sw  The changes have been merged to the master branch (i.e. for Rolling release). Does this fix your problems?

anaelle-sw, 2021-03-26:
Yes, I can build micro-ROS! Thanks a lot!

closed 2021-03-26 by anaelle-sw