ros2/rclc #75: API for removing a subscription

BrettRD, 2021-04-01:
There is no API for removing subscription, timers, services or clients. I ran into this while trying to add a context pointer to subscriptions.

`rclc_executor_add_subscription` increments `executor->index` after filling `executor->handles[executor->index]` with the information about the subscription.

During `rclc_executor_spin_some`, the executor iterates through all handles and makes a call to `rcl_wait_set_add_subscription`, then calls `rcl_wait`

If you call `rcl_subscription_fini` on the subscription, The RCL docs indicate `After calling, calls to rcl_wait and rcl_take will fail when using this subscription.`

No call in `executor.c` is able to decrement `executor->index` or remove a handle containing an invalid subscription.
Once you call rcl_subscription_fini, the executor breaks permanently.

I'd like to propose an API addition of:
`rcl_ret_t rclc_executor_remove_subscription(rclc_executor_t * executor, rcl_subscription_t * subscription);`
and similar for clients, timers and services.

This would search through `executor->handles` until it finds a pointer match to the subscription argument, remove it, and shorten the handles list.
I can't see a safe way of making this automatically detect dead subscriptions, especially in my case where my rcl_subscription_t are allocated dynamically.

Is the order of handles important?

JanStaschulat, 2021-04-01:
The order matters. An important feature compared to ROS 2 default executor, is, that the user can specify the execution order of callbacks. If multiple topics have arrived, then the callbacks will be called according to the order, in which the subscriptions were added to the executor. 

Yes, it is corect. We have not implemented the feature of removing the subscriptions. To problems have to be addressed:
- removing an subscription from the array
- copy all subscriptions one locations "to the left" in the array

Currently the executor-functions goes through the array until either the end of the array is reached, or the field is not valid. If you 
https://github.com/ros2/rclc/blob/2a8ec698d0f0e9be8e448dedb4ddc208435a009f/rclc/src/rclc/executor.c#L847

If you would simply remove on subscription, that those loops will be incorrect. This can be fixed easily. 

However, the total number of subscriptions is defined at initialization phase. With empty entries in the handles array, one either:
- would have to copy all elements after the removed subscription - to sustain a contiguous sequence of array fields again, `add_function` would then just append at the end of the array. 
- allow with empty entries inside the `handle_array` and implement a more advanced `add_function` which goes through the array and searches for the first empty entry and puts the new handle there.

BrettRD, 2021-04-01:
Implementing option B would require the addition of a flag to `rclc_executor_handle_t` to track if the handle has been invalidated, and require repeated tests for invalidated handles during spin, creating an on-going memory and performance penalty.

In option A, the handles are all shifted within statically allocated memory.
Ultimately this compiles down to a memmove operation within existing contiguous memory.
`rclc_executor_handle_t`s aren't all that heavy, and if I'm ever willing to wait for an operation it's when I'm adding or removing a subscription
This also means `spin_some()` and `scheduling()` iterate through the minimum number of handles every time.
A subscription that gets removed regularly is also likely to be nearly the last one, so I think the most common case is likely to be the best case (least data moved)
This makes it very easy to reason over the order of execution, because there's no insert operation to guess around.

I've implemented **option A**, and I think I got option A stack usage down to less than adding the callback.

BrettRD, 2021-06-30:
Implemented in https://github.com/ros2/rclc/pull/106

BrettRD, 2021-07-08:
Merged to Foxy

closed 2021-07-08 by BrettRD