ros2/rclc #79: Remove unecessary codecov token from repository secrets.

nuclearsandwich, 2021-04-16:
I've been reviewing ROS-related repositories which use Codecov after the [security incident](https://about.codecov.io/security-update/) announced yesterday. This repository uses the Codecov GitHub action and so was affected by the issue. The only secret exposed in the action is the configured codecov token and since the secret is not actually required I think the best course of action is to regenerate the codecov secret and remove it from the repository. I'm happy to take both operations but I wanted to notify the maintainers of the repository before doing so.

Versions of the codecov action >= 1.4.0 will now perform a checksum check of the bash script used internally to push data to codecov. Since this repository is using the action @v1 future actions runs will use an appropriate version of the codecov action. https://github.com/ros2/rclc/blob/905aaf94ac7dac2955cb179e342bf344ef235767/.github/workflows/ci.yml#L42

JanStaschulat, 2021-04-21:
I updated the CODECOV_TOKEN secret today.

nuclearsandwich, 2021-04-21:
Yep that's all that is necessary. Thanks!

closed 2021-04-21 by nuclearsandwich