ros2/rclc #86: Release rclc for Galactic

JanStaschulat, 2021-04-26:
update CI jobs for Galactic release.

BrettRD, 2021-04-27:
Leave https://github.com/ros2/rclc/pull/73 out; It doesn't change any existing API, and I'd like to backport as far as Foxy.
I also need to resubmit the PR because `rclc_executor_remove_subscription` needs to be a separate PR.

pablogs9, 2021-04-27:
I would like to have parameter server functionality in Galactic. When are we going to release @JanStaschulat?

JanStaschulat, 2021-04-27:
Galactic is going to be released on May 23 ([Galactic release plan](https://docs.ros.org/en/foxy/Releases/Release-Galactic-Geochelone.html)). I think, we should have the rclc release ready by then too. 

So I propose, one week before, May 14th to create a Galactic branch.

I apologize for this late notice. It's the first time we synchornize with a new ROS 2 release. Next time, we should align our timeline with the one from ROS 2 development.

ralph-lange, 2021-04-27:
That's still in time. No worries. To be on the safe side regarding the Release Candidate on May 17th, I propose to do the bloom already on May 11th or 12th because the PR to rosdistro may take more than a day.

ralph-lange, 2021-04-27:
> 
> 
> I would like to have parameter server functionality in Galactic. When are we going to release @JanStaschulat?

Yes, having parameters server functionality in Galactic would be really great!

JanStaschulat, 2021-04-27:
Proposed timeline:

05.05.2021 rclc repo: Function freeze and creation of Galactic branch
11.05.2021 rclc repo: Create bloom release (PR to rosdistro)
17.05.2021 ROS 2 Galactic: Release Candidate packages are built. Updated releases of ROS Desktop 2 packages available.
23.05.2021 ROS 2 Galactic: Announcement of Galactic release

JanStaschulat, 2021-04-27:
@pablogs9 Do you think this is enough time for you to merge the parameter PR until 5.5.2021?

pablogs9, 2021-04-27:
@JanStaschulat I'm not sure because we are having a lot of problems with message memory initiation because of `rosidl` runtimes and generators that use `malloc` and `free` instead of `rcutils_get_default_allocator`... Check https://github.com/ros2/rosidl/pull/584.

The point is that we are going to need to have a kind of conditional compilation depending on ROS2 or micro-ROS. But I'm thinking that maybe we can merge a basic (dynamic memory) version that freezes our parameter API and then makes changes internally taking care of API/ABI changes.

Let me work a little bit more on this and I will reach you with a solution.

JanStaschulat, 2021-04-27:
okay, no problem. If the API/ABI is the same, we can update the parameter support later on.

norro, 2021-04-28:
I would like to have the lifecycle services in galactic as they make lifecycle nodes so much more useful https://github.com/ros2/rclc/pull/51.

@pablogs9, @JanStaschulat Do you think we should/have to unstringify micro-ros lifecycle messages for the release (see @pablogs9's comment [here ](https://github.com/ros2/rclc/pull/51#discussion_r580059826))? This would probably involve changes in the agent, i.e. translation of micro-ros lifecycle messages _without_ strings to vanilla ROS 2 lifecycle messages.

pablogs9, 2021-04-28:
From the parameters, we have found that memory management in `rosidl_runtime_c` and `rosidl_generator_c` are not really prepared for micro-ROS because they rely on `malloc` directly. 

IMO the first step is to modify these two packages to use ROS 2 allocators (https://github.com/ros2/rosidl/pull/584) instead of raw dynamic memory functions. With these allocators, in micro-ROS, we will be able to customize the dynamic memory calls. But, in any case, dynamic memory operations should take place after the configuration step. 

The point is that those `rosidl_runtime_c` and `rosidl_generator_c` are integrated in the FastDDS typesupport ([here](https://github.com/ros2/rosidl_typesupport_fastrtps/blob/c6b4c93b7d475869a862c42095a5ec2f09f9dc21/rosidl_typesupport_fastrtps_c/resource/msg__type_support_c.cpp.em#L332)), so:
- that's means that FastDDS typesupport relies on **dynamic memory**.
- we cannot create high-level packages that init messages in static memory because when running on top of FastDDS middleware dynamic memory operations will cause crashes on that static memory.
- we cannot create micro-ROS high-level packages that rely on operation-time dynamic memory because this will harm realtimeness.

Regarding _unstringify_ micro-ROS packages, personally, I don't like it, because it forces users to launch or configure multiple things on the agent side. I would prefer to offer reduced functionality on the client side that works as expected without any other configuration step.

So, in summary:
- Once https://github.com/ros2/rosidl/pull/584 is merged and backported to foxy we will be able to control the dynamic memory operations in micro-ROS
- We should init strings in configuration steps using `rosidl_runtime_c` and `rosidl_generator_c` and ensure that have a default size: not only initting them, but assigning a certain amount of memory.
- When dealing with Micro XRCE typesupport these strings will not be reallocated in any case.
- When dealing with FastDDS typesupport there will be dynamic memory operations at runtime.

JanStaschulat, 2021-05-06:
I created a branch for "Galactic". 
https://github.com/ros2/rclc/tree/galactic
You may continue the development on the open pull request. After merging to the master branch, please also create a PR to backport it to "Galactic".

JanStaschulat, 2021-05-28:
Created new bloom release for Galactic: https://github.com/ros/rosdistro/pull/29702

closed 2021-07-05 by JanStaschulat