ros2/rclc #96: Create bloom release for Galactic

JanStaschulat, 2021-05-28:
galactic release 2.0.1

JanStaschulat, 2021-05-28:
bumped version to 2.0.1 https://github.com/ros2/rclc/pull/95
created pull request at rosdistro:  https://github.com/ros/rosdistro/pull/29702

JanStaschulat, 2021-05-28:
@pablogs9 @ralph-lange @norro We need to think carefully about tagging (for bloom releases). We have now different branches with the same major version 2.0.x (galactic, rolling). So it could happen, that we want to create a new release for rolling, say 2.0.1 but this tag has been assigned to galactic already! A tag can only be given once - branches are irrelevant. 

Maybe it is best to upgrade the tag for rolling to 3.0.x to avoid any version clashes in the future. What do you think?

pablogs9, 2021-05-28:
I'm ok with tagging 3.x.x the develop `master` branch

norro, 2021-05-28:
I usually create the tag `1.x.x` in the master branch (for rolling) and the according tags `1.x.x-foxy` etc in the according branches. In bloom, you can even provide this naming scheme for the distro tracks, so that it will remember for the next time.

I create the tags manually, though. I don't use bloom/catkin to create the tags fr me.

JanStaschulat, 2021-05-28:
I have to try it out, currently bloom release expects the same tag as the version in the package.xml. For example in package.xml the version is `2.0.1` and there must be a tag `2.0.1` How do I specify that bloom will accept `2.0.1-branchName` ?

norro, 2021-05-28:
`bloom-release --rosdistro <ros_distro> --track <ros_distro> <your_repository_name> --edit`
The `--edit` let's you configure the track and will ask you at one point, what tag to loom for. when answering this, you can use a wildcard for the version, something like `{version}-foxy` or similar.

JanStaschulat, 2021-05-28:
Cool, thanks.

closed 2021-05-31 by JanStaschulat