ros2/rclc #114: Provide a interface to initialize the executors memory before spinning

bjv-capra, 2021-07-09:
As a user of rclc, I'd like to be able to initialize all memory the executor will need before going into my "active" mode. Right now, this is done on the first `spin`. My current workaround is to call spin_some with a timeout of 0 at the end of my initialization. However, this is not fully transparent. "Why do you spin here?"

JanStaschulat, 2021-07-09:
Thanks for bringing this issue up, indeed a dedicated function, like `rclc_executor_prepare(rclc_executor_t * e)` could solve this issue. In this function all the setup for the wait_sets could be done. 

However, note, that also at runtime, one could add more handles (add_subscription, add_timer) etc. Then the wait_set will be updated (and memory allocation in rcl will happen). In that sense, whenever you update the objects the executor manages, then you will have "some" memory allocations again.

bjv-capra, 2021-07-09:
I agree with the second part completely, I think the behavior should be retained as is, is just to provide an extra step for the cases in which I would like all the memory to be preallocated before moving to the "runtime" phase. On embedded systems, I would completely avoid such behaviors as modifying the executor's objects during runtime. Nonetheless, there may be users that prefer/rely on this feature.
Thus, I'm simply asking to add the interface as you mentioned, `rclc_executor_prepare(rclc_executor_t *e)`

Thanks!

pablogs9, 2021-07-12:
Could you provide feedback of https://github.com/ros2/rclc/pull/118?

bjv-capra, 2021-07-12:
I'll check it out now, but won't be able to provide a thorough test.

The code would seem to do the trick though (I like how simple it was to adapt it)

bjv-capra, 2021-07-12:
@pablogs9 I tested it on my nodes, by using the prepared functionality, I don't have dynamic allocations done anymore during runtime. Excellent!

closed 2021-07-14 by pablogs9