ros2/rclc #115: Bug: client does not support callback with context

JanStaschulat, 2021-07-09:
Missing client callback with context:

@norro. It looks like that the implementation of the client callback with context is not properly called: it only calls the client callback without context:
https://github.com/ros2/rclc/blob/0263722fb48be9bddadef3601c50397c8d71a1d8/rclc/src/rclc/executor.c#L976
You never use a client callback with context?

BrettRD, 2021-07-09:
I don't think this is a bug

There is no way to add a client with a context pointer, and there is no typedef for a client callback to receive a context pointer.
https://github.com/ros2/rclc/blob/master/rclc/include/rclc/executor_handle.h#L130-L138

I think there is a subtle bug in the conditions to execute the callback (`WITH_CONTEXT` should throw an error instead of executing the callback), but not one that could cause a crash, and I can't see a way of even running that execution pathway.

This will disappear after https://github.com/ros2/rclc/issues/116
Maybe mark it as a feature request?

norro, 2021-07-20:
Agreed, the `WITH_CONTEXT` callback shouldn't be silently handled without context. Since I don't know any use case for a client with context, I suggest aborting with a warning/error, if a client callback is called `WITH_CONTEXT`.