ros2/rclc #116: Refactoring handle types

JanStaschulat, 2021-07-09:
Initally, we only had services, subscriptions, clients. Now we have multiple types of callbacks for subscriptions, services and clients:

- with request id/without request id (for services/clients),
- with context, without context (for subscriptions/clients/services).

It would make sense to refactor handle->type and add e..g
SERVICE_WITH_REQ_ID
SERVICE_WITHOUT_REQ_ID
SERVICE_WITHOUT_CONTEXT
SERVICE_WITH_CONTEXT
SUBSCRIPTION_WITH_CONTEXT
SUBSCRIPTION_WITHOUT_CONTEXT
same for CLIENTS ...
(quite a number of types ...)

the request-id is currently not used in services with context.

@norro @pablogs9 @BrettRD Let's discuss here, how to simplify callback-type and handle-type variables.

BrettRD, 2021-07-09:
I was quite surprised when the subscriptions had no context pointers, most C libraries I've used had context pointers everywhere to support `*this` pointers in cpp wrappers.
I would go so far as to suggest changing the API after Galactic to add context to all callbacks (reduce enum values and API size), and allowing NULL to be passed as a valid context pointer.

Foxy and Galactic refactor (No api changes) looks like it can be structurally very simple; every use of `handle->callback_type` is fully unrolled, and each switch statement is in a single case of `handle->type`, so the nested switch statements can be flattened, and `rclc_executor_handle_callback_type_t` can be deleted.

I would also like to rename  `rclc_callback_t` to `rclc_subscription_callback_t`, to make it more clear which types go with which `handle->type` enum value, but I would need to check how this can be done without breaking downstream application code.

JanStaschulat, 2021-07-09:
> Foxy and Galactic refactor (No api changes) looks like it can be structurally very simple; every use of `handle->callback_type` is fully unrolled, and each switch statement is in a single case of [...]

Hmm, req-id is something specific to services and clients. fully unrolling all possibilities (context with/without, req-id with/without) to (subscription, services, clients) sounds not so well structured to me. 

I'd rather keep a two-layered approach to distinguish between fundamental handle type and keep req-id as internal implementation detail: For example:
handle->type
- SUBSCRIPTION_DEFAULT
- SUBSCRIPTION_WITH_CONTEXT
- SERVICE_DEFAULT
- SERVICE_WITH_CONTEXT
-
callback_type
- CB_WITH_REQ_ID
- CB_WITHOUT_REQ_ID
-

BrettRD, 2021-07-10:
I don't mean to unfold all permutations of [no_args, context, req_id] * [service, client, gc, sub], most of those would be meaningless.

I mean create exactly as many handle->type enum values as we have callback types.

handle->type should be 1:1 with 
members of the function pointer union [8 of those](https://github.com/ros2/rclc/blob/foxy/rclc/include/rclc/executor_handle.h#L130-L138)
and API funcitons rclc_executor_add_<callback_type>, [8 of those too](https://github.com/ros2/rclc/blob/foxy/rclc/include/rclc/executor.h#L192-L435)
C types for callback function pointers, [7 of them](https://github.com/ros2/rclc/blob/foxy/rclc/include/rclc/executor_handle.h#L58-L88) because timer callbacks are defined elsewhere

[rclc_executor_handle_type_t](https://github.com/ros2/rclc/blob/foxy/rclc/include/rclc/executor_handle.h#L31-L39) becomes 
```
typedef enum
{
  SUBSCRIPTION,
//  SUBSCRIPTION_WITH_CONTEXT, // not currently in foxy 
  TIMER,
  CLIENT,
  CLIENT_WITH_REQUEST_ID,
  SERVICE,
  SERVICE_WITH_REQUEST_ID,
  SERVICE_WITH_CONTEXT,
  GUARD_CONDITION,
  NONE
} rclc_executor_handle_type_t;
```
and the callback_type can be removed completely

Then when executing the handle, there can be no invalid handle types. except the uninitialised value of [rclc_executor_handle_type_t::NONE](https://github.com/ros2/rclc/blob/c7e09de6c22f349ec456e7b966a3c907f516d894/rclc/include/rclc/executor_handle.h#L38)

The only place this gets slightly uglier are places like https://github.com/ros2/rclc/blob/foxy/rclc/src/rclc/executor.c#L765-L766 which become fall-through case statements
```
  switch (handle->type) {
    case SUBSCRIPTION:
    case SUBSCRIPTION_WITH_CONTEXT:
    {
      ...
    }
    break;
    case SERVICE:
    case SERVICE_WITH_REQUEST_ID:
    case SERVICE_WITH_CONTEXT:
    ...
```

JanStaschulat, 2021-07-14:
@BrettRD regarding your remark for subscription with default context pointer. We are currently discussing about C++ support for micro-ROS. In this scope, it definitly makes sense to re-think the rclc-API, e.g. have a context pointer as default. We are going to bring this issue up at today's embedded working group meeting, how this could be organized as a community effort.
 https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460. Please feel free to join.

pablogs9, 2021-07-14:
Next EWG is 27th of July

pablogs9, 2021-07-15:
Hello @JanStaschulat and @BrettRD,

are you taking into account to add context to `rclc_executor_add_timer(...)`?

IMO it is super interesting for example to do something like:

```c
void timer_callback(rcl_timer_t * timer, int64_t last_call_time,void * args)
{
	(void) last_call_time;
        rcl_publisher_t * publisher = (rcl_publisher_t *) args;
	rcl_publish(publisher, &msg, NULL));
}

void main(void){
...
    // create publisher
    rcl_publisher_t publisher;
....
	rcl_timer_t timer;
	const unsigned int timer_timeout = 1000;
	rclc_timer_init_default(
		&timer,
		&support,
		RCL_MS_TO_NS(timer_timeout),
		timer_callback);
	
	rclc_executor_add_timer(&executor, &timer, &publisher);
....
}
```

Currently, I usually have to define the publisher as global static in order to have access in the callback.

BrettRD, 2021-07-15:
@JanStaschulat I'll have to read up on the Arduino port, because that's all C++ at application level. I'll see you at the meeting!
@pablogs9 I'd love to see an arg/context/user pointer in timer callbacks. (and guard-conditions and clients)

I haven't read fully into the executor model and RCL, so I don't know why `timer` is provided by RCL and not RCLC.
If I do the refactor, I'll treat the timer as out-of-scope, but make obvious space for a smaller PR to include it.
It might be worth getting that far then raising an issue in RCL.

GLib / GStreamer never creates a function without a class pointer, and even then, adds another user pointer for convenience.
The ESP-ADF (audio subsystem) provides context pointers via a class member variables (hidden, but with getter function)

This API is in Foxy and Galactic now, so to avoid breaking downstream code, I'd add a bunch of `*_with_context` functions, and make another PR to prune it back in Rolling

It's possible to shim context onto the timers because the callback offers you a pointer you control.
```
// Don't do this for real.

// define a way of placing an additional object in memory immediately after a valid timer
typedef struct timer_with_args_t{
	rcl_timer_t timer;
	void * args;
};

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
	// cast the timer pointer to your wrapped type
	timer_with_args_t* wrapper = (timer_with_args_t*) timer;
	// retrieve the additional payload from the struct
	void* args = wrapper->args;
	(void) last_call_time;
        rcl_publisher_t * publisher = (rcl_publisher_t *) args;
	rcl_publish(publisher, &msg, NULL));
}

void main(void){
...
    // create publisher
    rcl_publisher_t publisher;
....
	// instead of instantiating just a timer, instantiate a timer and payload in memory together
	timer_with_args_t timer_with_args;
 	assert( &(timer_with_args.timer) == &timer_with_args );  //this is true by construction
	// get a type-correct pointer to the struct
	rcl_timer_t* timer = &(timer_with_args.timer);

	// put stuff into the payload
	timer_with_args.args = (void*) &publisher;

	const unsigned int timer_timeout = 1000;
	// init never refers to memory beyond the first part of the struct
	rclc_timer_init_default(
		timer,
		&support,
		RCL_MS_TO_NS(timer_timeout),
		timer_callback);
	// the executor will point to the timer_with_args we declared on the stack.
	rclc_executor_add_timer(&executor, timer);
....
}
```
I used a workaround like this to shim a context pointer onto the message pointer before the subscription-context PR.
This workaround is not possible on guard conditions.

If RCL wants to keep the code that triggers the callback, we can formalise the hack above as a type-inheritance model as offer that as the RCLC convenience wrapper; defining something like `typedef struct rclc_timer_t { rcl_timer_t timer; void* args};` and intercepting the callback to unpack it.  I'd prefer to see native support though.

JanStaschulat, 2021-07-15:
_"I haven't read fully into the executor model and RCL, so I don't know why timer is provided by RCL and not RCLC."_
@BrettRD I don't understand your comment. Could you elaborate?

Timers are available in rclc. You can initialize it with 
https://github.com/ros2/rclc/blob/master/rclc/include/rclc/timer.h
as you showed in your code example. 

In general, the rclc API does not introduce any new types, as rclcpp does, it consists only of a number of wrapper functions to simplify RCL calls. Executor is an exception. 

The same way, that we have defined service with context, subscription with context, why not have a timer with context?

BrettRD, 2021-07-16:
I just read into it, it's doable but it's a little squiggly.
My misunderstanding was based on the callback being called in [rcl timer](https://github.com/ros2/rcl/blob/foxy/rcl/src/rcl/timer.c#L285),
I simply hadn't noticed that the rcl timer call was triggered by [rclc executor](https://github.com/ros2/rclc/blob/foxy/rclc/src/rclc/executor.c#L1003)

`add_timer_with_context`, can be structured the same as `subscription_with_context`, using `handle->callback` and `handle->context`.
in `rclc_timer_init_default`, rclc would register its own function to unpack the user-args then call the user-callback held in `handle`.
At call time, rclc executor would instantiate an inherited timer type and and load `(rclc_executor_handle_t *) handle` as the payload. (`rcl_timer_h` only holds a pointer to a hidden struct, so a shallow copy is cheap)
inside the timer callback, `rcl_timer_h` is cast to `rclc_timer_h`,  `handle` is extracted, and `handle->callback(timer,handle->context)` gets called.

Staging this into Foxy and Galactic will be interesting.
I think a through refactor (Rolling) might consider removing the callback function pointer from https://github.com/ros2/rclc/blob/master/rclc/include/rclc/timer.h#L52

JanStaschulat, 2021-07-28:
@BrettRD thank you very much for your pull request. Merged it to Galactic and Rolling. I will close this ticket now.

closed 2021-07-28 by JanStaschulat