ros2/rclc #126: C++ support

JanStaschulat, 2021-07-15:
We want to discuss the requirements for a C++ API for micro-ROS. 
This could be 
- A) a replacement of rclcpp - without dynamic memory allocation
- B) thin layer on top of rclc

We are going to discuss different options and want to evaluate the effort in the next EWG meeting on July 27th. 
https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460

Please feel free to join and share your requirements.

JanStaschulat, 2021-07-27:
Feature comparison of future C++ API for micro-ROS `micro_rclcpp`. It consists of a number of C++ wrapper functions and uses all functions of RCLC library.  The table shows main differences to RCLCPP API and highlights restrictions and limitations:

| Feature        | rclcpp           | micro_rclcpp  |
| ---------------- |------------------| ------------------|
| Message type conversion      | Message types need to converted from C++ template type <br> argument to `rosid_typesupport_t` type in RCL. <br>Example for Subscription: <br> [rclcpp::create_subscription()](https://github.com/ros2/rclcpp/blob/1b28f389c28c7cd832c17599353808ac019f409d/rclcpp/include/rclcpp/create_subscription.hpp#L122) <br> [rclcpp::create_subscription_factory()](https://github.com/ros2/rclcpp/blob/1b28f389c28c7cd832c17599353808ac019f409d/rclcpp/include/rclcpp/subscription_factory.hpp#L110) <br> [rosidl_runtime_cpp::get_message_type_support_handle()](https://docs.ros2.org/foxy/api/rosidl_runtime_cpp/namespacerosidl__typesupport__cpp.html#a4282969ac1cf097f5688bb03cc7a3e3f)| similar approach can be used|
| | | |
| Callback function pointers     | no restrictions for function pointers <br> <br> <br><br> Message type in parameter declaration      |  Only [static member functions](https://stackoverflow.com/questions/19808054/convert-c-function-pointer-to-c-function-pointer) and [capture-less lamda](https://www.nextptr.com/tutorial/ta1188594113/passing-cplusplus-captureless-lambda-as-function-pointer-to-c-api). There is also an  [article about how use member functions as C-function pointers](http://p-nand-q.com/programming/cplusplus/using_member_functions_with_c_function_pointers.html) <br> only `void *` parameter for message type |
| | | |
| Message data types | staticly and dynamically allocated message variables (e.g. `std::String` and other STL containers)      |   dynamic memory allocation shall be avoided, `std::String` and STL-containers are not supported. This is not a restriction of `mirco_rclcpp` but a requirement of programming on a micro-controller |
|Node | Subscriptions, publishers, timers etc. are added to `Node` | RCLC does not have a `Node`concept. Subscriptions, publishers, timers are added directly to the Executor. One option would be to add a `Node` type in which the `Executor`is initialized and you can call `Node.spin()'  |

**Alternative:** 
Full C++ Implementation with Executor has the benefits:
- full support of callback function pointers and message type in function parameter

Limitations:
- no dynamicly allocated message types 

**Summary micro_rclcpp** 
Benefits:
- support for C++ API for Publishers, Subscribers, Timers, Services/Clients, guard conditions Nodes, real-time executor
- thin layer with C++ wrapper functions on top of RCLC
- 
Limitations:
 - callback function pointers: only static member functions, capture-less lambda expressions
 - callback function signature: no message types but only `void *` as argument possible
 - no dynamicly allocated message types supported (STL containers, e.g. `std::String`)
 -

JanStaschulat, 2021-07-27:
Questionnaire to micro-ROS users
- Why would you need a C++ API for micro-ROS?
- Are the limitations/restrictions described above okay for you?
- To which extend could you contribute (code development, reviewer, testing)?

I would like to ask the users of micro-ROS to give us some feedback from your perspective by answering the questionnaire. Thank you.

BrettRD, 2021-08-18:
This thread is looking a little sparse, I missed the meeting but I'll throw my opinion in

I've normally avoided C++ for microcontrollers because of how easy it is to accidentally pull in `std::` types that enthusiastically re-allocate memory.

Arduino in relies heavily on std::string and permits concatenation and deep copy; It's often ok even if it's not desired for performance.
If I were teaching with ROS on microcontrollers, I would appreciate an API that is more more congruent with the Arduino flavour.
A Node would allow some neat namespacing tools, and would make sense as a tidy top-level C++ object to handle entity destruction.

I would like to see more type-safety in RCLC; it's getting tedious writing endless functions that take only `void*` arguments. C++ templates are a good way of giving compile-time type-safety to the user.  I think a templated wrapper could lead to some more type-names being migrated back to RCLC, which would be good for readability.

I'd really **really** want to see strongly-typed callbacks, and I would make extensive use of `std::bind` to make *very* sure I can use callback context pointers to carry `this` pointers for non-static class member functions.  I would also make PRs to tweak callback argument order to make such hacks more seamless.

RCL offers allocation pointers and defined allocator structs, I'd personally default those to no-allocation, but I would want to see the custom-allocator feature of RCL exposed in micro_rclcpp as it is in rclcpp because of the zero-cost abstractions it offers.

Tangentially, I'm also interested in a Rust wrapper but that might be easier to build from scratch.

JanStaschulat, 2021-09-21:
Example for C++ Typesupport
- specifying in a subscription
- using ROS 2 type in callback

https://github.com/ros2/rclc/pull/199

bjv-capra, 2021-09-28:
Hi

I was not able to make my mic work during today's meeting, so I'll add my few cents here.
One has to avoid using almost all STL containers, `std::array` can be used with no risks. Even when exceptions are disabled, one can use at() with no more risks than the usual when working with arrays.
This added with all the benefits of type-safety, templates, and more. The overhead introduced by C++ can be substantial, however, from my point of view, it's none compared to the benefits.

We already work having wrappers on top of rclc or rcl depending on the use, and our biggest "struggle" is that we saw ourselves forced to make singletons to somehow get access to a class method during a callback (this was basically solved with the introduction of the context pointer).

Regarding RUST, it's interesting, but far from being mature enough to use it on product development.

JanStaschulat, 2021-09-28:
@bjv-capra Thank you for your feedback. Regarding the access to class members, as you mentioned, we have now subscriptions with context pointers. 

Would it be possible to share those wrappers as open source, so we could jointly provide a C++ API? 

Apart from access to class members (which should be solved), I see as main development points:
- typesupport (using rosidl_typesupport_handle functions from rclcpp and factoring out those on rclc)
- wrappers in Executor to call C++ type-safe callbacks from rclc (using context pointer) - as in the very simple ping-pong example https://github.com/ros2/rclc/blob/2c2287af3a4ed14eeeb50244a4faa8ebba20d8c7/rclc_examples/src/example_pingpong.cpp#L385