ros2/rclc #131: rclc_parameter: failing unit test master

JanStaschulat, 2021-07-16:
@pablogs9 
The github CI build job on Rolling fails: 
https://github.com/ros2/rclc/actions/runs/1036566413

This is the output of `colcon_build --packages-select rclc rclc_lifecycle rclc_parameter` on my local machine on Rolling:
```C
1: -- run_test.py: invoking following command in '/home/jst3si/github/build/rclc_parameter':
1:  - /home/jst3si/github/build/rclc_parameter/rclc_parameter_test --gtest_output=xml:/home/jst3si/github/build/rclc_parameter/test_results/rclc_parameter/rclc_parameter_test.gtest.xml
1: Running main() from /home/jst3si/ros2_rolling/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
1: [==========] Running 1 test from 1 test suite.
1: [----------] Global test environment set-up.
1: [----------] 1 test from Test
1: [ RUN      ] Test.rclc_node_init_default
1: 1626428338.597294 [0] rclc_param: using network interface wlp4s0 (udp/192.168.2.113) selected arbitrarily from: wlp4s0, br-b36361797d8d, br-14891b21ebd2, br-25a8035deb43, br-3a336396c85a, br-4785f765189b, br-89d99fc843c8, br-9b72c4be763f, br-adc9c608f11a, docker0, br-011ba5ddf9ec, br-1fa8df27d324, br-6bd062b25b91, br-5c6374d44867, br-815016a75d18, br-919ec2a858fb, br-997bb5b6c865, br-bcd6b201b56c, br-18ed396621d3, br-2e319e5f5bdf, br-89bec585106e, br-cf4299774085, br-de0a63c0f8fd, tun0
1: 
1: >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
1: This error state is being overwritten:
1: 
1:   'parameter is a null pointer, at /home/jst3si/github/src/rclc/rclc_parameter/src/rclc_parameter/parameter_server.c:235'
1: 
1: with this new error message:
1: 
1:   'node is a null pointer, at /home/jst3si/github/src/rclc/rclc_parameter/src/rclc_parameter/parameter_server.c:237'
1: 
1: rcutils_reset_error() should be called after error handling to avoid this.
1: <<<
1: rclc_parameter_test: /home/jst3si/ros2_rolling/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/TypeSupport2.hpp:405: virtual rmw_cyclonedds_cpp::TypedSpan<const char> rmw_cyclonedds_cpp::ROSIDLC_StringValueType::data(const void*) const: Assertion `str->capacity == str->size + 1' failed.
1: -- run_test.py: return code -6            
1: -- run_test.py: generate result file '/home/jst3si/github/build/rclc_parameter/test_results/rclc_parameter/rclc_parameter_test.gtest.xml' with failed test
1: -- run_test.py: verify result file '/home/jst3si/github/build/rclc_parameter/test_results/rclc_parameter/rclc_parameter_test.gtest.xml'
1/6 Test #1: rclc_parameter_test ..............***Failed    0.40 sec
...
83% tests passed, 1 tests failed out of 6

Label Time Summary:
cppcheck      =   0.62 sec*proc (1 test)
cpplint       =   0.76 sec*proc (1 test)
gtest         =   0.40 sec*proc (1 test)
lint_cmake    =   0.47 sec*proc (1 test)
linter        =   3.51 sec*proc (5 tests)
uncrustify    =   0.60 sec*proc (1 test)
xmllint       =   1.06 sec*proc (1 test)

Total Test time (real) =   3.91 sec

The following tests FAILED:
	  1 - rclc_parameter_test (Failed)
Errors while running CTest
--- stderr: rclc_parameter
Errors while running CTest
---
Finished <<< rclc_parameter [4.09s]	[ with test failures ]

Summary: 3 packages finished [23.8s]
  1 package had stderr output: rclc_parameter
  1 package had test failures: rclc_parameter
```

Maybe this has something to do with the recent PR regarding `set_compile_definition` macro in CMakeLists.txt?

JanStaschulat, 2021-07-16:
I checked on Galactic - if the old version still works, but the build failed due my older cmake version (below 3.12)
```C
CMake Error at CMakeLists.txt:8 (add_compile_definitions):
  Unknown CMake command "add_compile_definitions".

-- Configuring incomplete, errors occurred!
See also "/home/jst3si/github/build/rclc_parameter/CMakeFiles/CMakeOutput.log".
--- stderr: rclc_parameter                                                                          
CMake Error at CMakeLists.txt:8 (add_compile_definitions):
  Unknown CMake command "add_compile_definitions".

---
Failed   <<< rclc_parameter [2.47s, exited with code 1]
```

pablogs9, 2021-07-16:
Why are they passing in the ROS 2 build farm?

pablogs9, 2021-07-16:
@Acuadros95 can you check this?

JanStaschulat, 2021-07-16:
> Why are they passing in the ROS 2 build farm
good question. The CI job passed in the PR, but the default CI job on Rolling failed actually on `rclc_lifecycle`. So I triggered the build job again (maybe  some time out or so, I thought). But second run of the CI Rolling failed again. So I executed the `colcon_test` on my machine locally and got an error for `rclc_parameter`. Hmmm.

JanStaschulat, 2021-07-16:
```C
# ROS2  colcon
alias colcon_build="colcon build --symlink-install --cmake-args \" -DCMAKE_BUILD_TYPE=Debug\" --event-handler console_direct+ --base-path=src"
alias colcon_test="colcon test --base-path=src --event-handlers console_direct+ --pytest-args \" --capture=no\""
```

pablogs9, 2021-07-16:
I have found no problem replicating your procedure. Check my session here: https://asciinema.org/a/Q6PsZIUVMKKtdJVkXb8isCuQI

JanStaschulat, 2021-07-16:
Hmm, even example_executor fails: 
```
jst3si@RNGX5596:~/github/src/rclc/rclc_examples/src$ ros2 run  rclc_examples example_executor
1626433495.594680 [0] example_ex: using network interface wlp4s0 (udp/192.168.2.113) selected arbitrarily from: wlp4s0, br-b36361797d8d, br-14891b21ebd2, br-25a8035deb43, br-3a336396c85a, br-4785f765189b, br-89d99fc843c8, br-9b72c4be763f, br-adc9c608f11a, docker0, br-011ba5ddf9ec, br-1fa8df27d324, br-6bd062b25b91, br-5c6374d44867, br-815016a75d18, br-919ec2a858fb, br-997bb5b6c865, br-bcd6b201b56c, br-18ed396621d3, br-2e319e5f5bdf, br-89bec585106e, br-cf4299774085, br-de0a63c0f8fd, tun0
Created timer with timeout 1000 ms.
Created subscriber topic_0:
Debug: number of DDS handles: 2
example_executor: /home/jst3si/ros2_rolling/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/TypeSupport2.hpp:405: virtual rmw_cyclonedds_cpp::TypedSpan<const char> rmw_cyclonedds_cpp::ROSIDLC_StringValueType::data(const void*) const: Assertion `str->capacity == str->size + 1' failed.
Traceback (most recent call last):
  File "/home/jst3si/ros2_rolling/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli', 'console_scripts', 'ros2')())
  File "/home/jst3si/ros2_rolling/build/ros2cli/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/jst3si/ros2_rolling/build/ros2run/ros2run/command/run.py", line 70, in main
    return run_executable(path=path, argv=args.argv, prefix=prefix)
  File "/home/jst3si/ros2_rolling/build/ros2run/ros2run/api/__init__.py", line 73, in run_executable
    if -process.returncode in signal.valid_signals() and os.name == 'posix':
AttributeError: module 'signal' has no attribute 'valid_signals'
jst3si@RNGX5596:~/github/src/rclc/rclc_examples/src$ 
```

Difference is, I have a ROS 2 Rolling locally build on Ubuntu 18.04, which is not the default Ubuntu version (20.04). Pre-installed packages of ROS 2 Rolling are not available for Ubuntu 18.04.

JanStaschulat, 2021-07-17:
CI on Rolling is successful again 
https://github.com/ros2/rclc/actions/runs/1039767346

closed 2021-07-26 by JanStaschulat