ros2/rclc #171: Feature request: indicator that a best-effort publish completed

BrettRD, 2021-08-02:
I'm using a best-effort publisher in a fast loop, The non-blocking behaviour is great but I'm finding duplicate messages where the message was re-written by the data source before the publish completed.

The best-practices solution is multiple-buffering, where a pool of messages gets circulated through the data-writer and publisher, but I don't have any way of telling the pool when the message has been released by the publisher and can be re-used by the writer.

A callback (with a user-context pointer, released-message pointer, and optionally a `rcl_ret_t` return value) is an option, but might be overkill.

JanStaschulat, 2021-08-02:
> I'm using a best-effort publisher in a fast loop, The non-blocking behaviour is great but I'm finding duplicate messages where the message was re-written by the data source before the publish completed.

Is this an issue with micro-ros or does this error also occur on normal ROS 2 ?

BrettRD, 2021-08-03:
I don't have an easy way of testing multi-threaded behaviour from RCLC on Linux, except for the test suite in this repo.
I'm having trouble replicating what I'm seeing, I think it does rely on a significant serialisation delay.
https://github.com/BrettRD/rclc/blob/best-effort_safety_test/rclc/test/rclc/test_executor.cpp#L2641-L2713

In my application, I've declared the message on stack and I insert a pointer to one of a pool of 1kB DMA buffers into the message.
There's no memory allocation or copy operations, so the publisher thread can trigger two publish events very quickly on the same topic using the same message pointer.  I have timestamps for when every message gets packed and some of them are extremely close together.
I'm consistently receiving the correct number of messages, and for every duplicated sequence number, I have a gap in sequence number immediately *after* it.

I have a rotating pool of DMA buffers already, and I have plenty of ram to create a pool of metadata structures. I'll try that and see if it clears symptom.

BrettRD, 2021-08-03:
> I'm consistently receiving the correct number of messages, and for every duplicated sequence number, I have a gap in sequence number immediately _after_ it.

I misunderstood my problem.
If it were a new message overwriting the old, there should be a gap followed by a duplicate, I'm getting the reverse.
A pool of messages didn't fix it either.

I'm not sure what this bug is, or how to properly characterise it;
on *most* messages that have repeated sequence numbers, I get a duplicate timestamp, but there are messages with duplicated sequence that have different timestamps in a sensible order.

```
[INFO] [1627961766.331978403]: input_b seq is 58741760, stamp is 1627961766 sec, 344549000 ns, carrying 256 frames
[INFO] [1627961766.334033312]: input_b seq is 58742016, stamp is 1627961766 sec, 347757000 ns, carrying 256 frames
[INFO] [1627961766.344112972]: input_b seq is 58742272, stamp is 1627961766 sec, 357205000 ns, carrying 256 frames
[INFO] [1627961766.346061964]: input_b seq is 58742272, stamp is 1627961766 sec, 360065000 ns, carrying 256 frames
[WARN] [1627961766.346113911]: input_b buffer overlap, seq is 58742272, inserting 256 frames at a rewind of 256 frames,
[INFO] [1627961766.349319835]: input_b seq is 58742784, stamp is 1627961766 sec, 363978000 ns, carrying 256 frames
[INFO] [1627961766.349370276]: input_b lost samples,   seq is 58742784, skipped 256 frames
[INFO] [1627961766.352242153]: input_b seq is 58742784, stamp is 1627961766 sec, 363978000 ns, carrying 256 frames
[WARN] [1627961766.352292448]: input_b buffer overlap, seq is 58742784, inserting 256 frames at a rewind of 256 frames,
[INFO] [1627961766.361938258]: input_b seq is 58743296, stamp is 1627961766 sec, 375917000 ns, carrying 256 frames
[INFO] [1627961766.361984206]: input_b lost samples,   seq is 58743296, skipped 256 frames
[INFO] [1627961766.364114599]: input_b seq is 58743552, stamp is 1627961766 sec, 378774000 ns, carrying 256 frames
```

BrettRD, 2021-08-03:
I've added a subscription to the executor on the target device subscribing to what it sent, I'm seeing the same duplicate followed by gap on the target device, even without the 1K payload to slow it down, and even at a greatly reduce publish frequency.

I added an identical callback to sanity check the message contents immediately after publish, the outgoing message structures are clean and contiguous after publish is finished with them.

Is there a way I can cut the micro-ros agent out of this test and use an internal transport?

pablogs9, 2021-08-09:
With the default build, micro-ROS is not thread-safe. In order to enable this experimental feature check [this](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/4ccec07f4089ec1ec49e72c4ed0484ae6515cd75/CMakeLists.txt#L65).

BrettRD, 2021-08-09:
Is FreeRTOS on ESP32 a supported target for the multi-threaded client feature?
I'm getting a lot of errors for redefined int types and empty structs for mutex wrappers

pablogs9, 2021-08-09:
It should be supported, check [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/4ccec07f4089ec1ec49e72c4ed0484ae6515cd75/src/c/profile/multithread/multithread.c#L13)

Can you close this and open a new issue in the micro-ROS IDF component with the errors so we can check what is happening?

BrettRD, 2021-08-10:
Adding my own semaphores cleared up the message overwrite issue but created others.
continuing in https://github.com/micro-ROS/micro_ros_espidf_component/issues/81

closed 2021-08-10 by BrettRD