ros2/rclc #173: Server crashing after ros2 cli service call

norro, 2021-08-05:
## Issue template

- Hardware description: Ubuntu LTS vm
- Installation type: from source
- Version or commit hash: master (59632f29b)

#### Steps to reproduce the issue

1. Start example server - `ros2 run rclc_examples example_service_node`
2. Perform a ROS 2 CLI service call - `ros2 service call /addtwoints example_interfaces/srv/AddTwoInts "{a:24, b:42}"`

#### Expected behavior

Expected: you can perform the service call multiple times. This is still true, if you use `ros2 run rclc_examples example_client_node` instead of the CLI service call.

#### Actual behavior

example_service_node exits within 2-3 seconds.

#### Additional information

When I edit the example_service_node to print the rcl error:
```
77:  rcl_ret_t rc = rclc_executor_spin(&executor):
78:  printf("%d: %s", rc, rcl_get_error_string().str);
```
I get the following error printed 2-3 seconds aftre the CLI service call:
`601: rclc_executor_spin_some error, at [snip]/src/rclc/executor.c:1347`

pablogs9, 2021-08-09:
Is this happening with DDS middleware? or you are testing with XRCE-DDS?

norro, 2021-08-09:
Not yet with xrce-dds, this is happening with `rmw_cyclonedds_cpp`.

pablogs9, 2021-08-09:
I have been testing. Same issue here. Modifying the spin to a `rclc_executor_spin_some(...)` seems to not return.

Any idea @JanStaschulat?

pablogs9, 2021-08-09:
Have you tried with FastDDS RMW implementation?

norro, 2021-08-09:
@pablogs9 good catch! No issues when building/using it with `rmw_fastrtps_cpp`.

JanStaschulat, 2021-08-11:
I checked also at my rolling @ Ubuntu 18.04 installation. I had no problems. 
```
jst3si@RNGX5596:~$ ros2 service call /addtwoints example_interfaces/srv/AddTwoInts "{a: 24, b: 12}"
1628666194.897058 [0]       ros2: using network interface wlp4s0 (udp/192.168.2.113) selected arbitrarily from: wlp4s0, br-25a8035deb43, br-815016a75d18, docker0, br-997bb5b6c865, br-b36361797d8d, br-919ec2a858fb, br-011ba5ddf9ec, br-2e319e5f5bdf, br-3a336396c85a, br-4785f765189b, br-89d99fc843c8, br-1fa8df27d324, br-6bd062b25b91, br-bcd6b201b56c, br-de0a63c0f8fd, br-adc9c608f11a, br-cf4299774085, br-14891b21ebd2, br-18ed396621d3, br-5c6374d44867, br-89bec585106e, br-9b72c4be763f, tun0
requester: making request: example_interfaces.srv.AddTwoInts_Request(a=24, b=12)

response:
example_interfaces.srv.AddTwoInts_Response(sum=36)

jst3si@RNGX5596:~$ 
```

norro, 2021-08-11:
@JanStaschulat was that with cyclone dds?

norro, 2021-08-11:
@JanStaschulat Did you modify the example, see above:
```
77:  rcl_ret_t rc = rclc_executor_spin(&executor):
78:  printf("%d: %s", rc, rcl_get_error_string().str);
```
Then you should see, that with cyclone dds it is finishing after once request since it is raising the error stated above (otherwise failing silently).

JanStaschulat, 2021-08-11:
On my platform, the PR #175 solves this bug. @norro could you verify that it also works for you?

closed 2021-08-11 by JanStaschulat