ros2/rclc #198: Bug: Rolling: rclc_parameter

JanStaschulat, 2021-09-20:
Error in build job for rclc_parameter_test

```
/usr/bin/ld: CMakeFiles/rclc_parameter_test.dir/test/rclc_parameter/test_parameter.cpp.o: in function `rclcpp::topic_statistics::SubscriptionTopicStatistics<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> > >::publish_message_and_reset_measurements()':
  test_parameter.cpp:(.text._ZN6rclcpp16topic_statistics27SubscriptionTopicStatisticsIN14rcl_interfaces3msg15ParameterEvent_ISaIvEEEE38publish_message_and_reset_measurementsEv[_ZN6rclcpp16topic_statistics27SubscriptionTopicStatisticsIN14rcl_interfaces3msg15ParameterEvent_ISaIvEEEE38publish_message_and_reset_measurementsEv]+0x92): undefined reference to `rclcpp::Time::Time(long, rcl_clock_type_e)'
  /usr/bin/ld: CMakeFiles/rclc_parameter_test.dir/test/rclc_parameter/test_parameter.cpp.o: in function `rclcpp::topic_statistics::SubscriptionTopicStatistics<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> > >::SubscriptionTopicStatistics(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::Publisher<statistics_msgs::msg::MetricsMessage_<std::allocator<void> >, std::allocator<void> > >)':
  test_parameter.cpp:(.text._ZN6rclcpp16topic_statistics27SubscriptionTopicStatisticsIN14rcl_interfaces3msg15ParameterEvent_ISaIvEEEEC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt10shared_ptrINS_9PublisherIN15statistics_msgs3msg15MetricsMessage_IS5_EES5_EEE[_ZN6rclcpp16topic_statistics27SubscriptionTopicStatisticsIN14rcl_interfaces3msg15ParameterEvent_ISaIvEEEEC5ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt10shared_ptrINS_9PublisherIN15statistics_msgs3msg15MetricsMessage_IS5_EES5_EEE]+0x141): undefined reference to `rclcpp::Time::Time(long, rcl_clock_type_e)'
```
https://github.com/ros2/rclc/runs/3648788927?check_suite_focus=true
@norro @pablogs9 

Seems like rclcpp library is not found while linking.

JanStaschulat, 2021-09-20:
But there were no changes in rclc/time.h or rclcpp/src/time.cpp for a long time and Time::Time() constructor exists:

https://github.com/ros2/rclcpp/blob/2801553d61c5a30a0327d5cbc8d28bcd74e9703d/rclcpp/src/rclcpp/time.cpp#L60

JanStaschulat, 2021-09-20:
Any ideas?

pablogs9, 2021-09-20:
I cannot replicate it inside a Rolling docker. Do you know how to reproduce it?

```bash
root@f2affe60df17:/ws# tree -L 2
.
└── src
    └── rclc

2 directories, 0 files
root@f2affe60df17:/ws# colcon build --cmake-args -DBUILD_TESTING=ON
Starting >>> rclc    
Finished <<< rclc [7.44s]                     
Starting >>> rclc_lifecycle
Starting >>> rclc_parameter
Finished <<< rclc_lifecycle [6.89s]                                                                    
Finished <<< rclc_parameter [17.1s]                       
Starting >>> rclc_examples
Finished <<< rclc_examples [2.57s]                      

Summary: 4 packages finished [27.3s]
root@f2affe60df17:/ws# tree -L 2
.
├── build
│   ├── COLCON_IGNORE
│   ├── rclc
│   ├── rclc_examples
│   ├── rclc_lifecycle
│   └── rclc_parameter
├── install
│   ├── COLCON_IGNORE
│   ├── _local_setup_util_ps1.py
│   ├── _local_setup_util_sh.py
│   ├── local_setup.bash
│   ├── local_setup.ps1
│   ├── local_setup.sh
│   ├── local_setup.zsh
│   ├── rclc
│   ├── rclc_examples
│   ├── rclc_lifecycle
│   ├── rclc_parameter
│   ├── setup.bash
│   ├── setup.ps1
│   ├── setup.sh
│   └── setup.zsh
├── log
│   ├── COLCON_IGNORE
│   ├── build_2021-09-20_12-33-39
│   ├── latest -> latest_build
│   └── latest_build -> build_2021-09-20_12-33-39
└── src
    └── rclc

16 directories, 13 files
root@f2affe60df17:/ws#
```

JanStaschulat, 2021-09-21:
On `Linux 18.04 (bionic)` and using ROS 2 Rolling (built from source on July 9th 2021) with 
`export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`

Building and colcon test was successful:
```
colcon test --packages-up-to rclc_parameter
...
100% tests passed, 0 tests failed out of 6

Label Time Summary:
cppcheck      =   0.67 sec*proc (1 test)
cpplint       =   0.77 sec*proc (1 test)
gtest         =   0.66 sec*proc (1 test)
lint_cmake    =   0.50 sec*proc (1 test)
linter        =   3.71 sec*proc (5 tests)
uncrustify    =   0.75 sec*proc (1 test)
xmllint       =   1.01 sec*proc (1 test)

Total Test time (real) =   4.37 sec
Finished <<< rclc_parameter [4.58s]

Summary: 1 package finished [5.21s]
```

I also cannot reproduce the error.

JanStaschulat, 2021-09-21:
I am building `ROS 2 Rolling` locally on my Linux Ubuntu 18.04 laptop from sources again.

JanStaschulat, 2021-09-24:
The build job is running again https://github.com/ros2/rclc/actions/runs/1260672096

Something in the build chain at github/sync of ROS 2 Rolling distribution ???

closed 2021-09-24 by JanStaschulat