ros2/rclc #200: Microros Client reading operation

biontix, 2021-09-24:
Hello, I don't know if it's the right place to ask this question, but I didn't find any forum or group for that.

It is the function rclc_executor_spin_some() that go check if there is data available in the (custom)transport? I don't understand which is the flow to arrive to the Micro-XRCE-DDS function uxr_run_session_<...> and read the message just arrived. Like:

**RCLC**: rclc_executor_spin_some() >>  **RCL**: ... >> **RMW**: run_xrce_session() >> **Micro-XRCE-DDS-Client**: uxr_run_session()

In other word I would like to understand how (and who? maybe the executor?) MicroRos listen incoming message.
Thank you so much in advance.

pablogs9, 2021-09-27:
`rclc_executor_spin_some` -> `rcl_wait` -> `rmw_wait` -> `uxr_run_session_until_data` and the former will call your transport read and write functions.

In general the spin function in the executor is in charge of waiting for the middleware: in this step the middleware if ran and if there is any data it is stored in RMW layer static buffers, and the it performs a `rcl_take` that retrieves that data.

Let me know if you have further questions.

biontix, 2021-09-27:
Thank you so much! It really helped me.

closed 2021-09-27 by biontix