ros2/rclc #201: Feature request: Support string and list type parameters

maximiliankir, 2021-10-02:
For my application it would be nice to add string parameters and list parameters to the parameter server. For example, I would like to have topic names and frame_ids configurable via parameters. But for this I need string parameters. 
Is there already a feature which can be used for this? Is there a chance that this will be implemented in the near future? 
Or maybe someone can tell me what would have to be adapted for this, so that I can implement this extension myself.
Is there a reason why only float, int and boolean parameters were implemented, but strings and lists were omitted?

pablogs9, 2021-10-04:
Hello @maximiliankir, parameters have been implemented only for float, int, and boolean because they are fixed size. We decided that handling memory for strings and lists (not bounded) is out of the scope of a dynamic memory-aware implementation, but of course, we are open to contributions and we can discuss it.

maximiliankir, 2021-10-05:
I don't currently have enough time to implement this. I just thought there was already some possibility. As a workaround I will define a service for these configuration parameters. This is not very elegant, but it allows me to use more complex types.
Maybe I'll find the time someday, then I'll have another look at the implementation of the parameter server. I close the issue for now, when I work on it I' ll open it again.

closed 2021-10-05 by maximiliankir