ros2/rclc #223: Segfault in example_lifecycle_node

norro, 2021-12-13:
Calling `ros2 lifecycle get` to the current rolling example_lifecycle_node results in a segfault.
Calling `ros2 lifecycle set` to the current rolling example_lifecycle_node stalls.

norro, 2021-12-13:
This one is weird, as it keeps disappearing when I try to debug with gcc.

One suspicion of mine is that the context is not properly passed to the service, though. @JanStaschulat while looking at this, it seems that we don't have any tests for service with context?!

JanStaschulat, 2021-12-13:
@norro , there is a test case with service with context:
https://github.com/ros2/rclc/blob/27b4d4fc30149339da50376acd08893141748aa5/rclc/test/rclc/test_executor.cpp#L2566

JanStaschulat, 2021-12-13:
But indeed, the unit test does not check if the context was passed correctly:
https://github.com/ros2/rclc/blob/27b4d4fc30149339da50376acd08893141748aa5/rclc/test/rclc/test_executor.cpp#L2617

there it should have checked `srv1_id` (which is assigned the value of the context).
https://github.com/ros2/rclc/blob/27b4d4fc30149339da50376acd08893141748aa5/rclc/test/rclc/test_executor.cpp#L325

We need to update this unit test!

norro, 2021-12-13:
While this is true, this is not the root cause for the segfault.
I am rather passing a context that doesn't remain in scope.

JanStaschulat, 2021-12-13:
hmm, then I would look into the service callback function, if the a test for NULL pointer is done after the conversion of the context.

norro, 2021-12-13:
Was able to fix the segfault, get state and get available states services work as expected.

Change state service also works as expected when called via `ros2 service`, e.g., `ros2 service call /rclc/lifecycle_node/change_state lifecycle_msgs/ChangeState "{transition: {id: 1, label: configure}}"`. It still stalls, when called via `ros2 lifecycle set ...`, but this seems to be a different issue.

closed 2021-12-14 by norro