ros2/rclc #235: rclc_examples 3.0.5 failing to build on ROS 2 build farm

nuclearsandwich, 2022-01-17:
Since the rebuild on 2022-01-15 the `rclc_examples` package is no longer building.

* Package version: 3.0.5-1
* ROS 2 distribution: Rolling
* Platforms
  * [Ubuntu AMD64](https://build.ros2.org/job/Rbin_uF64__rclc_examples__ubuntu_focal_amd64__binary/) 
  * [Ubuntu ARM64](https://build.ros2.org/job/Rbin_ufv8_uFv8__rclc_examples__ubuntu_focal_arm64__binary/)

Since previous builds of `rclc_examples` version 3.0.5-1 [succeeded](https://build.ros2.org/job/Rbin_uF64__rclc_examples__ubuntu_focal_amd64__binary/170/) it is probable that this build failure is the result of changes in dependencies.

Here is an excerpt from the build log for [the most recent AMD64 failure](https://build.ros2.org/job/Rbin_uF64__rclc_examples__ubuntu_focal_amd64__binary/173):

```
[100%] Linking C executable example_action_server
/usr/bin/cmake -E cmake_link_script CMakeFiles/example_action_server.dir/link.txt --verbose=1
/usr/lib/ccache/cc -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-rolling-rclc-examples-3.0.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2   CMakeFiles/example_action_server.dir/src/example_action_server.c.o  -o example_action_server  -Wl,-rpath,/opt/ros/rolling/lib: /opt/ros/rolling/lib/librclc.so /opt/ros/rolling/lib/libexample_interfaces__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libexample_interfaces__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libexample_interfaces__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libexample_interfaces__rosidl_generator_py.so /opt/ros/rolling/lib/librcl_action.so /opt/ros/rolling/lib/librcl.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_generator_py.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_generator_c.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_c.so /opt/ros/rolling/lib/librcl_yaml_param_parser.so /opt/ros/rolling/lib/libyaml.so /opt/ros/rolling/lib/librmw_implementation.so /opt/ros/rolling/lib/librcl_logging_spdlog.so /opt/ros/rolling/lib/librcl_logging_interface.so /opt/ros/rolling/lib/libtracetools.so /opt/ros/rolling/lib/librmw.so /opt/ros/rolling/lib/librosidl_runtime_c.so /opt/ros/rolling/lib/libexample_interfaces__rosidl_generator_c.so /opt/ros/rolling/lib/libexample_interfaces__rosidl_typesupport_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libaction_msgs__rosidl_generator_py.so /opt/ros/rolling/lib/libaction_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/librosidl_typesupport_cpp.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_generator_py.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_generator_c.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_typesupport_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_generator_py.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/librosidl_typesupport_c.so /opt/ros/rolling/lib/librosidl_runtime_c.so /opt/ros/rolling/lib/librcpputils.so /opt/ros/rolling/lib/librcutils.so -ldl /usr/lib/x86_64-linux-gnu/libpython3.8.so -Wl,-rpath-link,/opt/ros/rolling/lib 
/usr/bin/ld: CMakeFiles/example_action_server.dir/src/example_action_server.c.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[4]: *** [CMakeFiles/example_action_server.dir/build.make:136: example_action_server] Error 1
make[4]: Leaving directory '/tmp/binarydeb/ros-rolling-rclc-examples-3.0.5/obj-x86_64-linux-gnu'
make[3]: Leaving directory '/tmp/binarydeb/ros-rolling-rclc-examples-3.0.5/obj-x86_64-linux-gnu'
make[3]: *** [CMakeFiles/Makefile2:425: CMakeFiles/example_action_server.dir/all] Error 2
make[2]: *** [Makefile:144: all] Error 2
```

Acuadros95, 2022-01-19:
I could not replicate this on our CI: https://github.com/ros2/rclc/pull/237
What are we missing @JanStaschulat ?

JanStaschulat, 2022-01-19:
@nuclearsandwich I am not aware of any changes we made in the repository. The last one was in December. The error message

15:14:09 /usr/bin/ld: CMakeFiles/example_action_server.dir/src/example_action_server.c.o: undefined reference to symbol 'pthread_create@@GLIBC_2.17'

maybe is due to some upstream dependency?

JanStaschulat, 2022-01-25:
created a new bloom release with a fix
https://github.com/ros/rosdistro/pull/31917

JanStaschulat, 2022-01-26:
PR merged.

closed 2022-01-26 by JanStaschulat