ros2/rclc #254: failing build on rolling

JanStaschulat, 2022-03-01:
CI job fails on rolling because Events Executor has been added to rmw (https://github.com/ros2/rmw/pull/286)

```
   In file included from /__w/rclc/rclc/ros_ws/src/rcl/rcl/include/rcl/client.h:27,
                   from /__w/rclc/rclc/ros_ws/src/rcl/rcl/src/rcl/client.c:20:
  /__w/rclc/rclc/ros_ws/src/rcl/rcl/include/rcl/event_callback.h:18:10: fatal error: rmw/event_callback_type.h: No such file or directory
     18 | #include "rmw/event_callback_type.h"
        |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
```
https://github.com/ros2/rclc/runs/5371188010?check_suite_focus=true
A new file (`event_callback_type.h`) has been added to the repository.
We cannot `sudo apt-get install` the new rmw package, because `rmw` has not been released as bloom release lately. 

https://github.com/ros/rosdistro/commits/master/rolling

@ralph-lange @pablogs9  Shall we clone repo in the CI action for now, like [here](https://github.com/micro-ROS/rmw_microxrcedds/blob/galactic/.github/workflows/ci.yml)?

ralph-lange, 2022-03-01:
Hi, @JanStaschulat, I like your idea of cloning the rmw repo in the CI action until it has been released.

pablogs9, 2022-03-01:
I'm also ok with cloning it

JanStaschulat, 2022-03-01:
But still I am wondering, why the most recent version of the  `rcl` package (with the EventExecutor PR which expects the header file in rmw) is downloaded in the CI action, but the `rmw` package is not .

JanStaschulat, 2022-03-01:
cloning the repo did not solve the problem
```
   -- Build files have been written to: /__w/rclc/rclc/ros_ws/build/rcl
  Scanning dependencies of target gtest
  [  0%] Building CXX object gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
  Scanning dependencies of target rcl
  [  1%] Building C object CMakeFiles/rcl.dir/src/rcl/arguments.c.o
  [  1%] Building C object CMakeFiles/rcl.dir/src/rcl/client.c.o
  In file included from /__w/rclc/rclc/ros_ws/src/rcl/rcl/include/rcl/client.h:27,
                   from /__w/rclc/rclc/ros_ws/src/rcl/rcl/src/rcl/client.c:20:
  /__w/rclc/rclc/ros_ws/src/rcl/rcl/include/rcl/event_callback.h:18:10: fatal error: rmw/event_callback_type.h: No such file or directory
     18 | #include "rmw/event_callback_type.h"
        |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  make[2]: *** [CMakeFiles/rcl.dir/build.make:76: CMakeFiles/rcl.dir/src/rcl/client.c.o] Error 1
  make[1]: *** [CMakeFiles/Makefile2:255: CMakeFiles/rcl.dir/all] Error 2
  make[1]: *** Waiting for unfinished jobs....
  [  2%] Linking CXX static library libgtest.a
  [  2%] Built target gtest
  make: *** [Makefile:141: all] Error 2
```
https://github.com/ros2/rclc/runs/5372137665?check_suite_focus=true
It still only builds rcl and does not take the local rmw.
 
I am trying now to add `rmw` as dependency in `package.xml`

JanStaschulat, 2022-03-01:
unfortunately they have not increased the version of rmw package. It is still 
Found rmw: 5.1.0 (/opt/ros/rolling/share/rmw/cmake)
(three month ago)

JanStaschulat, 2022-03-01:
clone the repo in a seperate step does not work, because `ros-tooling/action-ros-ci` creates it's own docker container and (for-whatever-reason) builds `rcl` but not `rmw` package.

christophebedard, 2022-03-01:
> Why is only rcl cloned and built in actions-ros-ci and not rmw? Could you include the rmw repository as dependency too?

This repo has `action-ros-ci` configured to clone `rcl` to build it from source:

* https://github.com/ros2/rclc/blob/master/dependencies.repos
* https://github.com/ros2/rclc/blob/2381ec52547c10e20881860509bc5a51e193931f/.github/workflows/ci.yml#L42

As I mentioned [over on the PR](https://github.com/ros2/rclc/pull/255), if you add `rmw` to the `dependencies.repos` file, it should fix this issue.

JanStaschulat, 2022-03-01:
@christophebedard thanks a lot for your suggestion. I added the relevant rmw repositories to [`dependencies.repos`](https://github.com/ros2/rclc/pull/255/files#diff-e0d946bdc4b7e275fb79d9a2d484fbe19fb84adf046001db89024be2463a903d). The PR now builds successfully :)

JanStaschulat, 2022-03-01:
Have to create an issue to remore these dependencies, once these changes in rcl/rmw become available as regular installation.

wjwwood, 2022-03-01:
We don't necessarily do releases after every change, so building just rcl from master while using rmw and other packages from binary is only going to work sometimes, because that's a combination we just don't test and honestly isn't a very logical selection of versions to build together. 

I'd recommend with building all dependencies from source or building none from source, but that sounds like something you can configure in the Actions you're using. 

An rmw release should happen in a few days.

JanStaschulat, 2022-03-02:
@wjwwood Thank you very much for your feedback.

closed 2022-03-02 by JanStaschulat