ros2/rclc #264: [rclc_lifecycle, galactic] running example_lifecycle_node failed.

ZhenshengLee, 2022-03-15:
## Issue template

- Hardware description: host
- RTOS: Ubuntu 20.04
- Installation type: apt
- Version or commit hash: galactic

#### Steps to reproduce the issue

```sh
# t1
source /opt/ros/galactic/setup.bash
zs@zs-3630:~/zs_ws/ros2_mcu$ ros2 run rclc_examples example_lifecycle_node 
creating lifecycle node...
registering lifecycle services...
registering callbacks...

# t2
zs@zs-3630:~/zs_ws/ros2_mcu$ ros2 service call /rclc/lifecycle_node/get_state lifecycle_msgs/srv/GetState 
requester: making request: lifecycle_msgs.srv.GetState_Request()

zs@zs-3630:~/zs_ws/ros2_mcu$ ros2 service call /rclc/lifecycle_node/get_state lifecycle_msgs/srv/GetState 
waiting for service to become available...
```

#### Expected behavior

ros service call can get result

#### Actual behavior

1. first service call cannot get result and stuck
2. example_lifecycle_node process exit after the service call

#### Additional information

using rmw_fastrtps_cpp

@pablogs9

pablogs9, 2022-03-15:
@JanStaschulat any idea?

JanStaschulat, 2022-03-15:
Are you configuring fastrtps as the DDS middlware in all terminals?
[`export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`](https://github.com/Ericsson/ros2-rmw_fastrtps)

ZhenshengLee, 2022-03-15:
> @ZhenshengLee Are you configuring fastrtps as the DDS middlware in all terminals? [`export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`](https://github.com/Ericsson/ros2-rmw_fastrtps)

Yes.

> #### Additional information
> using rmw_fastrtps_cpp

JanStaschulat, 2022-03-15:
@ZhenshengLee 
"example_lifecycle_node process exit after the service call" any error messages?

ZhenshengLee, 2022-03-15:
I also tested in microxrcedds with microROS-Agent (in host elmulator, not mcu)

the code is in https://github.com/ZhenshengLee/ros2_mcu/blob/master/app_ws/src/lifecycle_node/app.c

```sh
# t1
zs@zs-3630:~/zs_ws/ros2_mcu/agent_ws$ ros2 run micro_ros_agent micro_ros_agent udp4 -p 8888 -v6
[1647347961.593643] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1647347961.593810] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1647347972.899694] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 51 73 E2 2A 81 00 FC 01
[1647347972.899818] info     | Root.cpp           | create_client            | create                 | client_key: 0x5173E22A, session_id: 0x81
[1647347972.899863] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x5173E22A, address: 127.0.0.1:39844
[1647347972.899957] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1647347972.900114] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5173E22A, len: 52, data: 
0000: 81 80 00 00 01 07 2C 00 00 0A 00 01 01 03 00 00 1D 00 00 00 00 01 00 00 15 00 00 00 2F 72 63 6C
0020: 63 2F 6C 69 66 65 63 79 63 6C 65 5F 6E 6F 64 65 00 00 00 00
[1647347972.908097] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x5173E22A, participant_id: 0x000(1)
[1647347972.908154] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1647347972.908170] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1647347972.908251] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5173E22A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1647347972.908446] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5173E22A, len: 120, data: 
0000: 81 80 01 00 01 07 6E 00 00 0B 00 02 02 03 00 00 60 00 00 00 28 00 00 00 72 74 2F 72 63 6C 63 2F
0020: 6C 69 66 65 63 79 63 6C 65 5F 6E 6F 64 65 2F 74 72 61 6E 73 69 74 69 6F 6E 5F 65 76 65 6E 74 00
0040: 00 01 00 00 2C 00 00 00 6C 69 66 65 63 79 63 6C 65 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73
0060: 5F 3A 3A 54 72 61 6E 73 69 74 69 6F 6E 45 76 65 6E 74 5F 00 00 01 00 00
[1647347972.908515] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x5173E22A, topic_id: 0x000(2), participant_id: 0x000(1)
[1647347972.908542] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1647347972.908552] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1647347972.908598] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5173E22A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1647347972.908612] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5173E22A, len: 24, data: 
0000: 81 80 02 00 01 07 10 00 00 0C 00 03 03 03 00 00 02 00 00 00 00 00 00 01
[1647347972.908657] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x5173E22A, publisher_id: 0x000(3), participant_id: 0x000(1)
[1647347972.908679] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 03 00 00
[1647347972.908686] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1647347972.908697] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5173E22A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1647347972.908711] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5173E22A, len: 36, data: 
0000: 81 80 03 00 01 07 1C 00 00 0D 00 05 05 03 00 00 0E 00 00 00 00 02 01 00 03 00 01 00 0A 00 00 00
0020: 00 00 00 03
[1647347972.908870] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x5173E22A, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1647347972.908900] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 05 00 00
[1647347972.908913] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5173E22A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[... cut ...]
[1647347976.920298] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5173E22A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0F 00 00 00 80

# t2
zs@zs-3630:~/zs_ws/ros2_mcu/app_ws$ export RMW_IMPLEMENTATION=rmw_microxrcedds
zs@zs-3630:~/zs_ws/ros2_mcu/app_ws$ ros2 run micro_ros_lifecycle_node lifecycle_node 
creating lifecycle node...
[ERROR] [1647345731.603793383] []: Unable to initialize state machine: error not set, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/service.c:122
Error in creating lifecycle node.
```

ZhenshengLee, 2022-03-15:
> @ZhenshengLee "example_lifecycle_node process exit after the service call" any error messages?

No, there is no extra output in terminal.

> ```shell
> # t1
> source /opt/ros/galactic/setup.bash
> zs@zs-3630:~/zs_ws/ros2_mcu$ ros2 run rclc_examples example_lifecycle_node 
> creating lifecycle node...
> registering lifecycle services...
> registering callbacks...
> ```

JanStaschulat, 2022-03-15:
in terminal t1 you don't set fastrtps as DDS middleware.

pablogs9, 2022-03-15:
Related https://github.com/micro-ROS/micro-ROS-demos/issues/56

ZhenshengLee, 2022-03-16:
the core stack of debugging https://github.com/ZhenshengLee/ros2_mcu/blob/master/app_ws/src/lifecycle_node/app.c

after
```sh
ros2 service call /rclc/lifecycle_node/get_state lifecycle_msgs/srv/GetState 
```
the stack is

```sh
librclc_lifecycle.so!rclc_lifecycle_get_state_callback (Unknown Source:0)
librclc.so![Unknown/Just-In-Time compiled code] (Unknown Source:0)
librclc.so!rclc_executor_spin_some (Unknown Source:0)
librclc.so!rclc_executor_spin (Unknown Source:0)
main(int argc, const char ** argv) (/home/zs/zs_ws/ros2_mcu/app_ws/src/lifecycle_node/app.c:115)
```

JanStaschulat, 2022-03-16:
@ZhenshengLee, which code are you executing? In your app there is a [printf before calling spin](https://github.com/ZhenshengLee/ros2_mcu/blob/a1aee40a03f9159629829928792a996ed01d2810/app_ws/src/lifecycle_node/app.c#L114). But I don't see that in your terminal.

ZhenshengLee, 2022-03-16:
> @ZhenshengLee, which code are you executing? In your app there is a [printf before calling spin](https://github.com/ZhenshengLee/ros2_mcu/blob/a1aee40a03f9159629829928792a996ed01d2810/app_ws/src/lifecycle_node/app.c#L114). But I don't see that output in your terminal.

1. this issue is about codes from rclc_lifecycle galactic release via apt installation
2. in order to debug, I created https://github.com/ZhenshengLee/ros2_mcu/blob/master/app_ws/src/lifecycle_node/app.c and compile it in debug mode, and add a printf before calling spin. but this app.c is pretty similar with what in https://github.com/ros2/rclc/blob/master/rclc_examples/src/example_lifecycle_node.c
3. the call stack is from https://github.com/ZhenshengLee/ros2_mcu/blob/master/app_ws/src/lifecycle_node/app.c because I debug this program in my personally maintained repo.

JanStaschulat, 2023-02-03:
Galactic is EOL.

closed 2023-02-03 by JanStaschulat