ros2/rclc #271: Response to action server goal request

ted-miller, 2022-03-21:
- Hardware description: Yaskawa Motoman
- Version or commit hash: 3e5e219ebb5f2c7b3da74f5838349a7969676c81

#### Question

Is there currently a proper way to populate type-specific response data from an action server goal request? Is there a pointer somewhere to the response message? AFAICT, [the action response](https://github.com/ros2/rclc/blob/3e5e219ebb5f2c7b3da74f5838349a7969676c81/rclc/src/rclc/action_server.c#L82-L87) is a generic pass/fail with no additional data.

I've got a FollowJointTrajectory server implemented. If an invalid trajectory is submitted (or the robot isn't ready), then I want to reply to the user with information on WHY the trajectory was rejected. The [message format for the FJT action](https://github.com/ros-controls/control_msgs/blob/a555c37f1a3536bb452ea555c58fdd9344d87614/control_msgs/action/FollowJointTrajectory.action#L41-L49) allows for this additional information. But, I'm not seeing how to populate this.

Currently, I am simply returning `RCL_RET_ACTION_GOAL_REJECTED` from the `goal_callback` function. 

Any guidance would be appreciated.

pablogs9, 2022-03-21:
I have been checking the RCLCPP code for actions and when you reject a goal it is not possible to send any kind of data. For example, take a look at the generated type support of `FollowJointTrajectory` send goal response:

```cpp
template<class ContainerAllocator>
struct FollowJointTrajectory_SendGoal_Response_
{
...
  // field types and members
  using _accepted_type =
    bool;
  _accepted_type accepted;
  using _stamp_type =
    builtin_interfaces::msg::Time_<ContainerAllocator>;
  _stamp_type stamp;
...
}
```

The part of the `.action` file you point to is just the response for the GetResult service. That means that when the goal ends and the result is sent, this string can be filled.

My question now is, does the Action Client triggers the GetResult service when its goal is rejected? I'm going to take a look.

gavanderhoorn, 2022-03-21:
> That means that when the goal ends and the result is sent, this string can be filled.

aren't `CANCELLED` and `ABORTED` included in "goal processing end[ed]"?

Setting `error_code` and `error_string` as part of a goal rejection or abortion has been a standard pattern of usage for a long time.

Acuadros95, 2022-03-21:
Check ros2_controllers `FollowJointTrajectory` implementation goal callback: https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L909

The goal response does not use the `error_string`, but its used if the current goal is canceled: https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L1194

gavanderhoorn, 2022-03-21:
Almost irrelevant, but at least in ROS 1 `actionlib`, one could use [actionlib::ServerGoalHandle<...>::setRejected(const Result& result, const std::string& text)](https://docs.ros.org/en/melodic/api/actionlib/html/classactionlib_1_1ServerGoalHandle.html#af0a4d2debf5b19d985553903ddaef92c) like so:

```c++
control_msgs::FollowJointTrajectoryResult result;
result.error_code = control_msgs::FollowJointTrajectoryResult::INVALID_GOAL;
gh.setRejected(result, "This is not a valid goal for this server");
```

Could be the ROS 2 redesign lost that because it's using services for some part of the API.

Acuadros95, 2022-03-21:
You can check the logic behind ROS2 actions here: http://design.ros2.org/articles/actions.html, including differences between ROS 1 and ROS 2 actions

I agree with you, It looks like rejected goals are not very informative: `Rejected goals are not part of the state machine.` and there are no space for goal response on the .msg

gavanderhoorn, 2022-03-21:
So the work-around / expected implementation appears to be: accept (even though it's an invalid goal), then immediately cancel.

---

Looking at `rclcpp_action::GoalResponse`, perhaps returning `ACCEPT_AND_DEFER` and then immediately `ABORT`ing the goal would be 'cleaner'.

Still don't like it too much though, but that's not `rclc`'s fault.

ted-miller, 2022-03-21:
Thank you for the feedback. We'll figure something out on our end.

closed 2022-03-21 by ted-miller