ros2/rclc #273: rclc_parameter: Allow providing custom parameter descriptors

bjsowa, 2022-03-24:
The parameter descriptors returned by the `describe_parameters` service can include additional information like description of the parameter, read_only flag or value ranges. This information is, for example, utilized by [rqt_reconfigure](https://index.ros.org/p/rqt_reconfigure/). I think it would be a cool feature to allow providing custom descriptors for the parameters, similarly to how it is done in [rclcpp](https://docs.ros2.org/foxy/api/rclcpp/classrclcpp_1_1Node.html#a095ea977b26e7464d9371efea5f36c42).

pablogs9, 2022-03-25:
We are making a redesigning of the parameter server in micro-ROS right now. Our idea now is to divide the functionality into two modes a complete mode and a low memory mode because we found that the current implementation is really heap hungry.

Maybe we can integrate these descriptors in the complete mode CC: @Acuadros95 ?

Acuadros95, 2022-03-25:
No problem on adding this functionality, including read_only flag and value ranges.
> Maybe we can integrate these descriptors in the complete mode

Acuadros95, 2022-04-18:
Hi @bjsowa, we just added a lot of functionality on `rclc_parameter`: https://github.com/ros2/rclc/pull/274. 
Check it out! It included the describe parameters extra info.

closed 2022-04-19 by Acuadros95