ros2/rclc #278: Lifecycle library not built in micro-ROS static library builder

tnagyzambo, 2022-04-04:
- Hardware description: Arduino Portenta H7
- RTOS: mbedOS
- Installation type: platformio
- Version or commit hash: galactic

This is a continuation of the discussion [here](https://github.com/micro-ROS/micro_ros_arduino/issues/907).

It seems that the lifecycle library is not readily accessible in the Arduino micro-ROS environment in particular and to my understanding, MCU's in general.

Would it be possible to receive some clarification on this feature? Reading the micro-ROS website led me to understand that this was ready to experiment with.

JanStaschulat, 2022-04-04:
@pablogs9 could you check the specific scripts for Arduino on Galactic?

pablogs9, 2022-04-05:
I have been checking this and:
1. RCLC Lifecycle is built in the Arduino library, I didn't check it well yesterday: https://github.com/micro-ROS/micro_ros_arduino/tree/galactic/src/rclc_lifecycle
2. The error in https://github.com/micro-ROS/micro_ros_arduino/issues/907 is related to the linker not finding some functions, this is because the headers in `rclc_lifecycle` are not C++ ready and Arduino is C++. 
3. This PR adds conditional built of the `rclc_lifecycle` header in order to export the API with C mangling:https://github.com/ros2/rclc/pull/279

closed 2022-04-08 by pablogs9