ros2/rclc #300: Lifecycle service callback isuue

Cryoschrome, 2022-06-10:
- Hardware description: Teensy 3.6
- Firmware: Arduino
- Installation type: micro_ros_platformio
- Version or commit hash: humble

Hi, 
I am trying to create a lifecycle node using micro_ros_platformio. 
### Code (main.cpp)
```
#include <stdio.h>
#include <unistd.h>
#include <ArduinoJson.h>
#include <SPI.h>
#include <SD.h>
#include <TeensyStep.h>

#include <micro_ros_platformio.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rclc_parameter/rclc_parameter.h>
#include <rcl/error_handling.h>

#include "rclc_lifecycle/rclc_lifecycle.h"
#include <lifecycle_msgs/msg/transition_description.h>
#include <lifecycle_msgs/msg/transition_event.h>
#include <lifecycle_msgs/srv/change_state.h>
#include <lifecycle_msgs/srv/get_state.h>
#include <lifecycle_msgs/srv/get_available_states.h>
#include <lifecycle_msgs/srv/get_available_transitions.h>

#include <rcl_interfaces/msg/log.h>

//define motor pins
#define STEP    2
#define DIR     3
#define ENABLE  4
#define ALARM   5
#define BRAKE   6

#define RCCHECK(fn)     { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){errorLoop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void errorLoop()
{
  while (1) {
    digitalWrite(13, !digitalRead(13));
    delay(100);
  }
}

rcl_node_t node;
rclc_lifecycle_node_t lifecycleNode;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;
rclc_parameter_server_t parameterServer;

static const char * jointID          = "j0001";

//create motor control objects
Stepper motor(STEP, DIR);
StepControl stepMotor;
RotateControl rotateMotor;

rcl_ret_t onJointConfigure()
{
  for(int i=0; i<4; i++)
  {
    digitalWrite(13, !digitalRead(13));
    delay(500);
  }
  return RCL_RET_OK;
}

rcl_ret_t onJointActivate()
{
  digitalWrite(13, HIGH);
  return RCL_RET_OK;
}

rcl_ret_t onJointDeactivate()
{
  digitalWrite(13, LOW);
  return RCL_RET_OK;
}

rcl_ret_t onJointCleanup()
{
  for(int i=0; i<6; i++)
  {
    digitalWrite(13, !digitalRead(13));
    delay(500);
  }
  return RCL_RET_OK;
}

void setup() {
  Serial.begin(115200);
  pinMode(13, OUTPUT);

  set_microros_serial_transports(Serial);
  delay(2000);
  allocator = rcl_get_default_allocator();
  rclc_support_init(&support, 0, NULL, &allocator);

  //Initialise node and parameter server
  RCCHECK(rclc_node_init_default(&node, jointID, "", &support));
  //Make the node to lifecycle node
  rcl_lifecycle_state_machine_t stateMachine_ = rcl_lifecycle_get_zero_initialized_state_machine();
  RCCHECK(rclc_make_node_a_lifecycle_node(
    &lifecycleNode,
    &node,
    &stateMachine_,
    &allocator,
    true
  ));

  //initialise executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 6, &allocator));
  rclc_lifecycle_service_context_t context;
  context.lifecycle_node = &lifecycleNode;
  RCCHECK(rclc_lifecycle_init_get_state_server(&context, &executor));
  RCCHECK(rclc_lifecycle_init_get_available_states_server(&context, &executor));
  RCCHECK(rclc_lifecycle_init_change_state_server(&context, &executor));

  RCCHECK(rclc_lifecycle_register_on_configure(&lifecycleNode, &onJointConfigure));
  RCCHECK(rclc_lifecycle_register_on_activate(&lifecycleNode, &onJointActivate));
  RCCHECK(rclc_lifecycle_register_on_deactivate(&lifecycleNode, &onJointDeactivate));
  RCCHECK(rclc_lifecycle_register_on_cleanup(&lifecycleNode, &onJointCleanup));
}

void loop() {
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```
### colcon_lowmem.meta
```
{
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=5",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=6",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
```
The callback for the lifecycle services does not get called. 
### Output
```
└─➞ ros2 service call /j0001/get_state lifecycle_msgs/srv/GetState {}
requester: making request: lifecycle_msgs.srv.GetState_Request()
```

#### Expected behavior
Response is given is by the server

#### Actual behavior
No response is given by the server

Any help is appreciated
Thank you!

JanStaschulat, 2022-06-14:
does normal publisher and subscriber work on your platform? 
CC @pablogs9 do you have this hardware available and could run this test?

pablogs9, 2022-06-14:
Some points:
1. `rclc_lifecycle_service_context_t context;` is in the `setup()` stack and probably overwritten when `loop()` stack starts. Is this context important in the lifecycle workflow?
2. Could you share the log of the micro-ROS Agent with the flag `-v6`?

JanStaschulat, 2022-06-14:
Compared to the [lifecycle example ](https://github.com/ros2/rclc/blob/master/rclc_examples/src/example_lifecycle_node.c) the context must be available when you call `spin_some`.

So try using global variables for context, stateMachine_ etc , so they are visible in `setup()` and `loop()`

pablogs9, 2022-06-14:
Thanks, @JanStaschulat. Please @Cryoschrome could you try and report? If it still does not work, I will try to replicate it here.

Cryoschrome, 2022-06-14:
Thanks @pablogs9 and @JanStaschulat!
Will makes the changes as per the points you mentioned and report back ASAP!

Cryoschrome, 2022-06-15:
So, I have made both `context` and `stateMachine_` global, not the code goes into the `errorLoop()` when executing `rclc_make_node_a_lifecucle_node()`. 
I have provided the updated code and agent log below.
### Updated Code
```
#include <stdio.h>
#include <unistd.h>
#include <ArduinoJson.h>
#include <SPI.h>
#include <SD.h>
#include <TeensyStep.h>

#include <micro_ros_platformio.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rclc_parameter/rclc_parameter.h>
#include <rcl/error_handling.h>

#include "rclc_lifecycle/rclc_lifecycle.h"
#include <lifecycle_msgs/msg/transition_description.h>
#include <lifecycle_msgs/msg/transition_event.h>
#include <lifecycle_msgs/srv/change_state.h>
#include <lifecycle_msgs/srv/get_state.h>
#include <lifecycle_msgs/srv/get_available_states.h>
#include <lifecycle_msgs/srv/get_available_transitions.h>

#include <rcl_interfaces/msg/log.h>

//define motor pins
#define STEP    2
#define DIR     3
#define ENABLE  4
#define ALARM   5
#define BRAKE   6

#define RCCHECK(fn)     { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){errorLoop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void errorLoop()
{
  while (1) {
    digitalWrite(13, !digitalRead(13));
    delay(100);
  }
}

rcl_node_t node;
rclc_lifecycle_node_t lifecycleNode;
rclc_lifecycle_service_context_t context;
rcl_lifecycle_state_machine_t stateMachine_;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;
rclc_parameter_server_t parameterServer;

static const char * jointID = "j0001";

//create motor control objects
Stepper motor(STEP, DIR);
StepControl stepMotor;
RotateControl rotateMotor;

rcl_ret_t onJointConfigure()
{
  for(int i=0; i<4; i++)
  {
    digitalWrite(13, !digitalRead(13));
    delay(500);
  }
  return RCL_RET_OK;
}
[... 144 lines cut ...]
[1655276901.107566] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x6460B8CA, len: 12, data: 
0000: 81 80 07 00 03 01 04 00 00 12 00 02
[1655276901.107614] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x6460B8CA, object_id: 0x0000
[1655276901.107679] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x6460B8CA, len: 14, data: 
0000: 81 80 06 00 05 01 06 00 00 12 00 02 00 00
[1655276901.107695] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x6460B8CA, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 08 00 00 00 80
[1655276901.112716] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x6460B8CA, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 07 00 00 00 80
[1655276901.112764] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x6460B8CA, len: 20, data: 
0000: 81 80 08 00 03 01 04 00 00 13 00 05 03 01 04 00 00 14 00 03
[1655276901.113062] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x6460B8CA, object_id: 0x0000
[1655276901.113114] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x6460B8CA, object_id: 0x0000
[1655276901.113147] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x6460B8CA, len: 14, data: 
0000: 81 80 07 00 05 01 06 00 00 13 00 05 00 00
[1655276901.113168] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x6460B8CA, len: 14, data: 
0000: 81 80 08 00 05 01 06 00 00 14 00 03 00 00
[1655276901.113182] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x6460B8CA, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 09 00 00 00 80
[1655276901.117559] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x6460B8CA, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 08 00 00 00 80
[1655276901.117590] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x6460B8CA, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 09 00 00 00 80

```

pablogs9, 2022-06-15:
Your micro-ROS client is only creating a `lifecycle_msgs/TransitionEvent` publisher and a `lifecycle_msgs/ChangeState` service and then is failing.

Maybe this is related to the colcon meta configuration issue with platform.io: https://github.com/micro-ROS/micro_ros_platformio/issues/33

In order to check this, could you find `rmw_microxrcedds_c/config.h` in your include folder and check if `#define RMW_UXRCE_MAX_SERVICES 6` is correctly set?

Cryoschrome, 2022-06-15:
I check the `rmw_microxrcedds_c/config.h` and no `#define RMW_UXRCE_MAX_SERVICES` is not set correctly. Would updating it to the correct value in `rmw_microxrcedds_c/config.h` fix the issue?

pablogs9, 2022-06-15:
Ok, the colcon metas are misbehaving.

You need to create a user meta (instead of modifying the provided colcon_lowmem.meta) as explained here: https://github.com/micro-ROS/micro_ros_platformio#other-configuration

In order to make this "user meta" work, please use this branch: https://github.com/micro-ROS/micro_ros_platformio/pull/34

Just note that this branch is not working in platformio 6.0.2, so make sure that you have an older version.

Also make sure that you start in a clean environment, with no micro-ROS built.

Sorry for this workaround, but we are just working in this fix right now. If you could provide feedback it would be nice.

pablogs9, 2022-06-20:
Any update on this?

pablogs9, 2022-06-23:
Which is the output of `ros2 service list`?

Cryoschrome, 2022-06-23:
`ros2 service list` gives the following output.
```
/node/change_state
/node/get_available_states
/node/get_available_transitions
/node/get_state
/node/get_transition_graph

```

pablogs9, 2022-06-23:
Service is up & running. @JanStaschulat have you tested lifecycle in micro-ROS w/ Micro XRCE-DDS rmw? is the inner memory for types being correctly initialized and so on?

JanStaschulat, 2022-06-23:
@norro Has lifecycle been tested in micro-ROS with Micro XRCE-DDS rmw?

olneumann, 2025-06-01:
@JanStaschulat @pablogs9 Is there an update to this issue? I tried with a Teensy 4.1 board today and still face the issue that I get stuck in the error loop. I checked the meta configuration for the amount of services and is correct. In the agent dump I get the message:

`[1655276901.113114] debug    | ProxyClient.cpp    | delete_object_unlock     | object deleted         | client_key: 0x6460B8CA, object_id: 0x0000`