ros2/rclc #312: rclc and rclcpp avoiding python usage in cross-compilation

RocaPiedra, 2022-08-29:
Hi,

I am trying to use ROS 2 on a target by doing cross-compilation (Xilinx UltraScale +). I want to avoid the usage of Python and the possibility of using C is very interesting. I would like to know if it would be possible to use rclc and rclcpp to create static libraries and compile the packages and avoid the need for Python in the target.

Thanks in advance

pablogs9, 2022-09-06:
Hello @RocaPiedra, it is possible to cross-compile the ROS 2 stack in static libraries and executables for targetting embedded applications. One approach is using [micro-ROS](https://micro.ros.org/) if you have an embedded bare-metal or RTOS platform. But if you have a full-featured OS, it should be possible to build the entire ROS 2 stack (including RCLC or RCLCPP) in static libraries.

RocaPiedra, 2022-09-06:
Hi @pablogs9!

The OS is PetaLinux (based in Yocto), so from what I've checked, it doesn't seem like micro-ROS is compatible (but I've seen it is possible to compile micro-ROS in a host, so maybe I could do the same workflow and cross-compile with the aarch64 compiler?). I thought about building the stack directly but haven't found any guidance on the topic and I thought there was some kind of dependency with Python. If you have any available information or references on how to compile it to static libraries (and use it), it would help a lot!

JanStaschulat, 2022-09-07:
Hi @RocaPiedra, why do you think that micro-ROS is not compatible with PetaLinux? As you mentioned, micro-ROS can be used on a Linux host. micro-ROS was developed for micro-controllers, which have only very little resources (CPU, memory). But it can also be used on larger platforms. 

As an alternative for cross-compiling  micro-ROS  with the server-client architecture (server as Linux host, client the micro-controller), you could also just cross-compile rclc. Then the interface to ROS2 stack on your Yocto Linux is rclc (e.g. C-API). It depends on the system resources of the hardware (memory and CPU). 

The rclc layer provides a programming interface for C applications but can also be used in C++ programs (with some restrictions, e.g. static member functions as callback functions). But you could use the C-API also directly from your C++ program.  See also this issue [Feature request: C++ supportS](https://github.com/ros2/rclc/issues/126).

RocaPiedra, 2022-09-07:
Hi @JanStaschulat, the HW I use could run ROS 2 without much problem but I haven't been able to install ROS yet. I'm updating the PetaLinux version (from 2018.3 to 2020) to include [meta-ros](https://github.com/ros/meta-ros) but my base image is not working and I have not found a Yocto recipe for the version I have now (rocko 2.4). 
So the host approach in Linux (without cross compilation) is not an option because I haven't been able to install ROS 2 so far. That is why I'm trying to find an option where I don't need to use Python at all and just depend on C/C++ so I can load whatever I need on the target and compile it. I guess from @pablogs9's answer that I could do that but without guidance, I feel like it will take me a very long time.

alecGraves, 2023-04-02:
I would also like to build ROS2 without Python. Ideally, I would like to build core components (like RCL and RCLC and a DDS backend) without CMake or a C++ compiler either.

I think there is great value in reducing the number of dependencies. That makes ros2 more portable and accessible, including on new or niche platforms like custom processors and operating systems.

By my testing, RCL requires 78 packages in the workspace just to compile (without testing enabled), or 103 if you count colcon extensions. 48 of these packages (ament + colcon) are just related to the build system. Having a minimal  backup build system like Makefiles for core components like RCL and a minimal DDS backend would help to significantly reduce bloat in the build system and improve portability.