ros2/rclc #325: microros rclc executor allows 5 handles but not 6

bryangd34, 2023-02-02:
Hi everyone,

- Hardware description: Pico from RaspberryPi
- Version: Foxy

I originally had this posted on ROSanswers [here](https://answers.ros.org/question/412111/microros-rclc-executor-allows-5-handles-but-not-6/). I was advised to post the issue here also.

Does the microros rclc executor have a maximum allowed number of handles that can be added? I couldn't find anything documented about this, so perhaps I have some other kind of memory issue (or something else altogether).

I was able to run an executor with 4 publishers + 1 subscriber, but the moment I add another publisher, there is no data on the new published topic. I then tried to split the publishers and subscribers over two executors. The behaviour was still the same. A snippet of my code:

   `rclc_executor_t executor_1, executor_2;
    executor_1 = rclc_executor_get_zero_initialized_executor();
    executor_2 = rclc_executor_get_zero_initialized_executor();
    rclc_executor_init(&executor_1, &support.context, 4, &allocator);
    rclc_executor_init(&executor_2, &support.context, 2, &allocator);
    rclc_executor_add_timer(&executor_1, &publisher1_timer);
    rclc_executor_add_timer(&executor_1, &publisher2_timer);
    rclc_executor_add_timer(&executor_1, &publisher3_timer);
    rclc_executor_add_subscription(&executor_1, &subscriber, &sub_msg, &subscription_callback, ON_NEW_DATA);

    rclc_executor_add_timer(&executor_2, &publisher4_timer);
    rclc_executor_add_timer(&executor_2, &publisher5_time);

    rclc_executor_prepare(&executor_1);
    rclc_executor_prepare(&executor_2);

    while (true)
    {
            val1 = val1_read();
            val2 = val2_read();
            val3 = val3_read();
            rclc_executor_spin_some(&executor_1, RCL_MS_TO_NS(1000));
            val4 = val4_read();
            val5 = val5_read();
            rclc_executor_spin_some(&executor_2, RCL_MS_TO_NS(1000));
    }

    return 0;`

Acuadros95, 2023-02-02:
Hi @bryangd34,

The issue here is that the default number of timers is set to 4 with the RMW configuration parameter `RMW_UXRCE_MAX_GUARD_CONDITION`.
Increase this value on your `colcon.meta` configuration and rebuild your library.

Check this tutorial for more details: [Memory management tutorial](https://docs.vulcanexus.org/en/humble/rst/tutorials/micro/memory_management/memory_management.html#entity-creation)

bryangd34, 2023-02-02:
Hi @Acuadros95 
Thank you. I have checked my `colcon.meta` and I added that parameter (it wasn't there by default) and set it to 5. Following this I rebuilt the library.
Unfortunately the topic still does not have any data being published to it. If I remove one of the other publishers, then this topic _does_ have data in it. Here is my colcon.meta file:
`{
    "names":{
        "microxrcedds_client":{
            "cmake-args":[
                "-DBUILD_SHARED_LIBS=ON"
            ]
        },
        "microcdr":{
            "cmake-args":[
                "-DBUILD_SHARED_LIBS=ON"
            ]
        },
        "rosidl_typesupport_microxrcedds_c":{
            "cmake-args":[
                "-DBUILD_SHARED_LIBS=ON"
            ]
        },
        "rosidl_typesupport_microxrcedds_cpp":{
            "cmake-args":[
                "-DBUILD_SHARED_LIBS=ON"
            ]
        },
        "rcutils":{
            "cmake-args":[
                "-DENABLE_TESTING=ON"
            ]
        },
        "rmw_microxrcedds":{
            "cmake-args":[
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=127.0.0.1",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=8888",
                "-DRMW_UXRCE_MAX_NODES=3",
                "-DRMW_UXRCE_MAX_PUBLISHERS=5",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=5",
                "-DRMW_UXRCE_MAX_CLIENTS=5",
                "-DRMW_UXRCE_STREAM_HISTORY=32",
                "-DRMW_UXRCE_MAX_HISTORY=10",
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=1000",
                "-DRMW_UXRCE_MAX_GUARD_CONDITION=5"
            ]
        }
    }
}
`
Do you have any other suggestions I could try out please?

Acuadros95, 2023-02-02:
How did you rebuild the library? Can you try with the parameter set on a clean build?

To make sure that the change is applied, you can check the defined value under the library include `build/rmw_microxrcedds/include/rmw_microxrcedds_c/config.h`:

```c
#define RMW_UXRCE_MAX_GUARD_CONDITION 5
```

You should also check the return values from the executor methods, they will return an error code if the timer could not be included on the spin.
And as another suggestion, try to make it work with a single executor. They are not limited on their number of handles.

bryangd34, 2023-02-02:
Thanks. I just did `colcon build`. So, you make a good point there. I will try rebuild clean and also confirm the parameter is set.

I'll also try to check the return values from the executor methods. I have just now updated the code to have a single executor.

I will update shortly here once I have tested. Thanks again.

bryangd34, 2023-02-02:
I did a clean build. Then I checked `build/rmw_microxrcedds/include/rmw_microxrcedds_c/config.h` and strangely the value is not updated.

`#define RMW_UXRCE_MAX_GUARD_CONDITION 4`

I don't really get why it is not updating. Now this seems like more of a `colcon` issue. I'm trying to search for this. But if you have any suggestions, please let me know. Thanks.

JanStaschulat, 2023-03-01:
@bryangd34 Is this issue resolved?

bryangd34, 2023-03-01:
@JanStaschulat 
Unfortunately not. I got stuck at the point mentioned in my previous message, that is 
`define RMW_UXRCE_MAX_GUARD_CONDITION 4`
does not get updated after build.
I am proceeding with just 5 handles for the time being. But would like to at some point increase it to 6.
Thanks.

Acuadros95, 2023-03-01:
@bryangd34 Could you detail which build system are you using? How did you start a clean build and where is your `colcon.meta`?

This is related to colcon args or your build process, should be straightforward to fix.

bryangd34, 2023-03-01:
Hi @Acuadros95 
I am building on a RaspberryPi with Ubuntu 20 and Foxy installed.
Clean build: I removed the build, log and install directories. Then ran `colcon build`.
My `colcon.meta` is in micro_ros_setup, that is `micro_ros_setup/colcon.meta`

Acuadros95, 2023-03-02:
You need to use the `--metas` args with colcon build to set the directory where your `colcon.meta` file is located.

Example: `colcon build --metas micro_ros_setup`

JanStaschulat, 2023-04-21:
@bryangd34 Is this issue resolved now?

bryangd34, 2023-04-21:
@JanStaschulat unfortunately not.

This has become a lower priority for me, hence my lack of updates. My apologies.

The latest is, after deleting the microros workspace and starting again from scratch, I built the workspace again using: `colcon build --metas micro_ros_setup` . Strangely now, after a build without errors, I don't have the directory `build/rmw_microxrcedds` , which was certainly there previously as I had to check `build/rmw_microxrcedds/include/rmw_microxrcedds_c/config.h`.

So I am stuck at this point. I'll gladly try out any suggestions you may have. Thanks!

flabrosse, 2024-11-19:
Same problem here.  I had 5 subscriptions, timers and services, and need to add a new service.  `rclc_service_init_default` returns an error (`RCL_RET_ERROR`) when initialising the 6th.  I have added the line
```
                "-DRMW_UXRCE_MAX_GUARD_CONDITION=6"
```
to `./components/micro_ros_espidf_component/colcon.meta`  (it was not there before) and the file `components/micro_ros_espidf_component/micro_ros_src/build/rmw_microxrcedds/include/rmw_microxrcedds_c/config.h` now has the appropriate `define` after a clean/build.

I still get the error on the 6th service.

flabrosse, 2024-11-19:
I just spotted the variable `-DRMW_UXRCE_MAX_SERVICES=1` in `colcon.meta`.  This is clearly ignored as I had 3 services working fine.  Where else can this be changed?

flabrosse, 2024-11-19:
Solved, for me.  Here's how, so that it could be useful to someone else.  I had a `app-colcon.meta` file in my top directory.  Edited that one, cleaned, built and it is all fine now.

bryangd34, 2025-05-12:
Closing this issue. The original issue was never solved and I have long since moved onto other projects. But now faced a similar problem which was solved by editing `colcon.meta`.

closed 2025-05-12 by bryangd34