ros2/rclc #340: [rclc] Executor for multi-threaded and non-POSIX operating systems (AUTOSAR)

JanStaschulat, 2023-03-01:
Extend Executor to support
- multi-threaded execution with priority assignment to subscription callbacks
- non-POSIX operating systems, aka no threads but periodic task activation

Tasks:

- [x] [baseline branch](https://github.com/ros2/rclc/tree/feature/restructured-executor)
- [ ] Executor prototype (only timer, subscription, limited functionality) for single-threaded, multi-threaded
- [ ] Executor prototype for non-POSIX OS
- [ ] Discussion and Feedback
- [ ] Feature complete Executor (all other handles and functions)

see current status of pull request [rclc#339](https://github.com/ros2/rclc/pull/339)

Zard-C, 2023-10-07:
Hi Jan, this work has captured my keen interest, could you show  some relevant discussions and information?

tulpe1977, 2023-10-13:
Hi @Zard-C , early this year I have been thinking about a better software archicture that would support single-threaded, multi-threaded and non-POSIX executor. Here are the slides about it: 
[2023-02-design-document-modular-micro-ROS-Executors.pdf](https://github.com/ros2/rclc/files/12893216/2023-02-design-document-modular-micro-ROS-Executors.pdf)

I don't have a written document about it yet. But we could have a discussion on micro-ROS slack channel or reach out via email (jan.staschulat@de.bosch.com).

JanStaschulat, 2023-10-13:
Hi @Zard-C , early this year I have been thinking about a better software archicture that would support single-threaded, multi-threaded and non-POSIX executor. Here are the slides about it:
[2023-02-design-document-modular-micro-ROS-Executors.pdf](https://github.com/ros2/rclc/files/12893216/2023-02-design-document-modular-micro-ROS-Executors.pdf)

I don't have a written document about it unfortunately, but I think the slides represent the concept quite well. If you have questions/suggestions/remarks, please reach out on micro-ROS slack channel or via email ([jan.staschulat@de.bosch.com](mailto:jan.staschulat@de.bosch.com)).

Zard-C, 2023-10-13:
> Hi @Zard-C , early this year I have been thinking about a better software archicture that would support single-threaded, multi-threaded and non-POSIX executor. Here are the slides about it: [...]

Much appreciated!