ros2/rclc #344: Requesting release of rclc in Rolling and Humble

cottsay, 2023-03-20:
Hi, it's been a while since this package has been released. #314 was merged about 2 months ago and was backported to humble. I'm normally not so invested in getting fixes like this released, but we're trying to roll out a change on the buildfarm that will break this package without that fix, which I'd prefer not to do.

A release in Rolling and Humble would be appreciated.

pablogs9, 2023-03-21:
Can you take a look @JanStaschulat ?

JanStaschulat, 2023-03-22:
Bloom release PR:
- Humble: https://github.com/ros/rosdistro/pull/36563
- Rolling: https://github.com/ros/rosdistro/pull/36564

JanStaschulat, 2023-03-28:
Re-run bloom-release for Rolling https://github.com/ros/rosdistro/pull/36620

closed 2023-03-30 by JanStaschulat