ros2/rclc #349: Merge this into rcl?

alecGraves, 2023-04-01:
Would we want to merge the work done on the rclc executor back into RCL at any point? I think a lot of other clients might want the ability to define sequential execution or perform priority-based event handling. This is a very common need in robotic systems.

Is there a point to keeping this separate from the RCL?

JanStaschulat, 2023-04-04:
Hi @alecGraves thanks for your question. (CC @fujitatomoya @pablogs9 )
Some arguments that support rclc as a seperate package on top of RCL:

- The ROS2 stack is structured in layers: RMW, RCL, language frontends like RCLCPP, RCLPY, and RCLC. We consider rclc rather as a language frontend, rather than a core library that is used by other frontends. 
- integrating rclc into rcl would lead to the situation having two executors in two layers, one in RCL, one in RCLCPP, which are using the same underlying rcl interface (rcl_wait, rcl_take, rcl_publish). Therefore, I guess, this would be more confusing than simplifying the design. 
- "I think a lot of other clients might want the ability to define sequential execution or perform priority-based event handling." Yes, I fully agree. But what is the drawback in installing the rclc package, which provides the rclc-Executor with the mentioned features, in your use-case?

alecGraves, 2023-04-13:
For my project, I need a C API. I would like to minimize build dependencies, and having 78 repositories for core functionality makes porting to a new platform more difficult.

I see your point that ros2 is a layered architecture, and I did not know that RCL did not have any Executor implementation. Thanks.

closed 2023-04-13 by alecGraves