ros2/rclc #350: Subscriber receives incorrect message

vtran5, 2023-04-14:
## Issue template

- Hardware description: PC
- RTOS: Ubuntu 22.04
- Installation type: Per instruction in [the rclc README.md](https://github.com/ros2/rclc/blob/master/README.md)
- Version or commit hash: rclc: master latest, ros2: humble

#### Steps to reproduce the issue
The code is in the rclc_app package in this [repo](https://github.com/vtran5/thesis/blob/main/rclc/rclc_app/src/my_app.c). It is a modification of the example_executor.c but with 1 extra publisher-subscriber pair. The second publisher is called inside the first subscriber callback.
It can be built and run the same as the rclc_examples but with a dependency in [custom_interfaces package](https://github.com/vtran5/thesis/tree/main/custom_interfaces).
#### Expected behavior
The message received in the second subscriber callback should be the same as the published message in the first subscriber callback.
#### Actual behavior
The message received in the second subscriber callback is different.
#### Additional information

JanStaschulat, 2023-04-17:
Hi @vtran5, the code looks correct to me. Could you post the contents of the send/received messages to understand the difference between the receveived message?

JanStaschulat, 2023-04-17:
Just a comment: 
> Version or commit hash: rclc: master latest, ros2: humble
this will not solve the problem, but if you are using ROS2 Humble, you should also use rclc version Humble (and not master. The master branch is equivalent to ROS 2 Rolling).

vtran5, 2023-04-17:
@JanStaschulat Thank you for your comment. I have found the issue with my code as I give a pointer to pointer of the message to the rcl_publish instead of a pointer. I will close the issue.

closed 2023-04-17 by vtran5