ros2/rclc #366: CI jobs on ROS2 build farm are not triggered for Iron and Rolling

JanStaschulat, 2023-06-14:
@pablogs9 @Acuadros95
the ci pipeline on OSRF build farm is not triggered for Rolling and Iron for all of the pending pull requests. Do you have any idea what else to check?

I have also seen that you have updated the line in one of the micro-ros/setup repos:
https://github.com/ros2/rclc/blob/master/.github/workflows/ci.yml#L26

- uses: actions/checkout@v2 to
- uses: actions/checkout@v3

might this be the reason? Where is this documented? Could you provide a link to the documentation for the changes of v2 and v3?

pull requests: 

- https://github.com/ros2/rclc/pull/363
- https://github.com/ros2/rclc/pull/362
- https://github.com/ros2/rclc/pull/359
- https://github.com/ros2/rclc/pull/358
- https://github.com/ros2/rclc/pull/357

Acuadros95, 2023-06-14:
I would say its related to the branch used, maybe something needs to be configured on the build farm side?

JanStaschulat, 2023-06-14:
I updated the action/checkout version for some PR. This did not resolve the issue.

JanStaschulat, 2023-06-14:
> I would say its related to the branch used, maybe something needs to be configured on the build farm side?

That's a good point.

JanStaschulat, 2023-06-14:
Indeed, Iron and Rolling build jobs are using the `master` branch:

https://github.com/ros/rosdistro/blob/0ad9c9e92cab4d0955ebf35dd3237acabf7b38a4/iron/distribution.yaml#L4068
https://github.com/ros/rosdistro/blob/0ad9c9e92cab4d0955ebf35dd3237acabf7b38a4/rolling/distribution.yaml#L3967

I need to change the branch to rolling and iron, respectivly.

JanStaschulat, 2023-06-14:
I would do the following: 
- I would merge all the pending pull requests. The Github Action CI job ran successfully or is on hold (for one PR)
- then I have a clean status on master, rolling and iron.
- then I can create a new revision number, add a tag
- then create a new bloom release (change the branch names for Iron and Rolling). 

@Acuadros95 , @pablogs9 Do you agree?

JanStaschulat, 2023-06-15:
Created a bloom release for Iron, however the `iron` branch was not updated with the `bloom-release` tool. I need to create another pull reqest again that changes the branch name. 
PR on ros/rosdistro: https://github.com/ros/rosdistro/pull/37664

closed 2023-06-21 by JanStaschulat