ros2/rclc #378: Better executor type names

JanStaschulat, 2023-06-27:
Having recently updated this package in my local build env to include this PR, I started running into build errors.

One of my platform headers already `#define`s a `NONE`, leading to problems with the value in the `rclc_executor_type_t` `enum` ([here](https://github.com/ros2/rclc/pull/355/files#diff-2b97ef21c9b9eb498bff3a8723efb6469f26889ffd0ab32e5b6e88ca2594abcaR53-R60)).

While I would agree with a statement along the lines of "that's not a very good idea to globally `#define` something called `NONE`", unfortunately I can't change my platform's header. 

Would it perhaps be possible to prefix the names in the `enum` in `executor.h`, such that the chances of clashing with existing/platform `#define`s are reduced? Perhaps an `EXECUTOR_` prefix?

---

Edit: the `LET` member in `rclc_executor_semantics_t` is also perhaps a candidate for a prefix, although I haven't yet ran into an issue with that.

_Originally posted by @gavanderhoorn in https://github.com/ros2/rclc/issues/355#issuecomment-1605652975_

gavanderhoorn, 2023-06-27:
The suggested `NON_INITIALIZED_EXECUTOR` (for `NONE)` and `LOGICAL_EXECUTION_TIME` (for `LET`) would solve the immediate issue I have (ie: the name clash).

As I wrote  in https://github.com/ros2/rclc/pull/355#issuecomment-1607218381, some of the other members (`SINGLE_THREADED` and `MULTI_THREADED`) are rather generically named as well.

Seeing as there is no enum scope in C11, prefixing those might also be a good idea.

gavanderhoorn, 2023-06-27:
I'll close this with #379 and #382 merged.

gavanderhoorn, 2023-06-27:
O wait, it's your issue @JanStaschulat. Can't close it.

closed 2023-06-27 by JanStaschulat