ros2/rclc #385: rclc_example: memory leak

Zard-C, 2023-07-28:
## A memory leak occurs when running the rclc_example

- Hardware description:   x86_64 GNU/Linux
- RTOS: 
- Installation type: 
- Version or commit hash:  humble

#### Steps to reproduce the issue

```shell
source install/setup.zsh 
valgrind --tool=memcheck --leak-check=full ./install/rclc_examples/lib/rclc_examples/example_sub_context 
```

#### Expected behavior
no memory leaking 

#### Actual behavior
```shell
==115904== Memcheck, a memory error detector
==115904== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==115904== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==115904== Command: ./install/rclc_examples/lib/rclc_examples/example_sub_context
==115904== 
Created subscriber topic_foo:
Created subscriber topic_bar:
Created subscriber topic_baz:
Debug: number of DDS handles: 3

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 0

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 100

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 300

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 1

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 101

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 301

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 2

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 102

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 302

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 3

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 103

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 303

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 4

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 104

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 304

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 5

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 105

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 305

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 6

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 106

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 306

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 7

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 107

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 307

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 8

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 108

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 308

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 9

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 109

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 309
==115904== 
==115904== HEAP SUMMARY:
==115904==     in use at exit: 42,654 bytes in 97 blocks
==115904==   total heap usage: 12,916 allocs, 12,819 frees, 3,748,842 bytes allocated
==115904== 
==115904== 3 bytes in 3 blocks are definitely lost in loss record 1 of 23
==115904==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==115904==    by 0x48E9006: __default_allocate (allocator.c:37)
==115904==    by 0x4BCEF61: rosidl_runtime_c__String__init (string_functions.c:34)
==115904==    by 0x48D6030: std_msgs__msg__String__init (string__functions.c:25)
==115904==    by 0x109B61: main (example_sub_context.c:112)
==115904== 
==115904== LEAK SUMMARY:
==115904==    definitely lost: 3 bytes in 3 blocks
==115904==    indirectly lost: 0 bytes in 0 blocks
==115904==      possibly lost: 0 bytes in 0 blocks
==115904==    still reachable: 42,651 bytes in 94 blocks
==115904==         suppressed: 0 bytes in 0 blocks
==115904== Reachable blocks (those to which a pointer was found) are not shown.
==115904== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==115904== 
==115904== For lists of detected and suppressed errors, rerun with: -s
==115904== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```
#### Additional information

JanStaschulat, 2023-07-28:
@Zard-C thanks for this issue and the fix. Could you run valgrind with the fixed version and, thereby, document here that there is no memory leak any more?

Zard-C, 2023-07-28:
> @Zard-C thanks for this issue and the fix. Could you run valgrind with the fixed version and, thereby, document that the memory leak was resolved?
Hi, here is the  valgrind report of the fixed version,
```shell
==143265== Memcheck, a memory error detector
==143265== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==143265== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==143265== Command: ./install/rclc_examples/lib/rclc_examples/example_sub_context
==143265== 
Created subscriber topic_foo:
Created subscriber topic_bar:
Created subscriber topic_baz:
Debug: number of DDS handles: 3

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 0

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 100

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 300

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 1

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 101

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 301

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 2

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 102

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 302

Published message Hello World! on topic_foo
Callback: I heard: Hello World! on topic_foo
Callback: context contains: foo counting from zero
Callback: context also contains: 3

Published message Hello World! on topic_bar
Callback: I heard: Hello World! on topic_bar
Callback: context contains: bar counting from 100
Callback: context also contains: 103

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
[... 85 lines cut ...]
Callback: context contains: bar counting from 100
Callback: context also contains: 109

Published message Hello World! on topic_baz
Callback: I heard: Hello World! on topic_baz
Callback: context contains: baz counting from 300
Callback: context also contains: 309
==143265== 
==143265== HEAP SUMMARY:
==143265==     in use at exit: 42,651 bytes in 94 blocks
==143265==   total heap usage: 14,486 allocs, 14,392 frees, 3,879,879 bytes allocated
==143265== 
==143265== LEAK SUMMARY:
==143265==    definitely lost: 0 bytes in 0 blocks
==143265==    indirectly lost: 0 bytes in 0 blocks
==143265==      possibly lost: 0 bytes in 0 blocks
==143265==    still reachable: 42,651 bytes in 94 blocks
==143265==         suppressed: 0 bytes in 0 blocks
==143265== Reachable blocks (those to which a pointer was found) are not shown.
==143265== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==143265== 
==143265== For lists of detected and suppressed errors, rerun with: -s
==143265== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

```

Zard-C, 2023-07-28:
This issue is fixed by https://github.com/ros2/rclc/pull/386

PR #386 (2023-07-28): fix rclc_example: memory leaking in msg.data allocation
Fix bug: memory leaking while msg.data allocation.

This bug is caused by using malloc allocate memory for the std_msg.data.data, but we call std_msgs__msg__String__init(), msg.data.data was allocated a block memory which size is 1, thus we should use allocator.reallocate to reallocate memory for msg.data.data. 

Fixes #385
closed 2023-07-28 by Zard-C