ros2/rclc #393: "'rcl_timer_init2’ was not declared in this scope" error during initial Colcon build; Build failed

NikolaZgonjanin, 2023-09-19:
- Hardware description: Personal Computer (VM)
- Installation type: Cloned from GitHub
- Version or commit hash: Humble (7af2054)

#### Steps to reproduce

1. Install ROS 2 Humble following the [official DOCS](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html)
2. Source the ROS distribution
3. Make a new workspace (ros_ws; In example rclc_ws)
4. Build using Colcon
5. Git Clone the repository into the workspace
6. Build using Colcon

#### Expected behavior

The workspace builds with no errors

#### Actual behavior

After entering `colcon build` command for the second time, the error log bellow is shown and the build fails

```shell
nikola@local:~/rclc_ws$ colcon build
Starting >>> rclc    
--- stderr: rclc                             
/home/nikola/rclc_ws/rclc/rclc/src/rclc/timer.c: In function ‘rclc_timer_init_default2’:
/home/nikola/rclc_ws/rclc/rclc/src/rclc/timer.c:36:18: warning: implicit declaration of function ‘rcl_timer_init2’; did you mean ‘rcl_timer_init’? [-Wimplicit-function-declaration]
   36 |   rcl_ret_t rc = rcl_timer_init2(
      |                  ^~~~~~~~~~~~~~~
      |                  rcl_timer_init
/home/nikola/rclc_ws/rclc/rclc/test/rclc/test_executor.cpp: In member function ‘virtual void TestDefaultExecutor::SetUp()’:
/home/nikola/rclc_ws/rclc/rclc/test/rclc/test_executor.cpp:575:7: error: ‘rcl_timer_init2’ was not declared in this scope; did you mean ‘rcl_timer_init’?
  575 |       rcl_timer_init2(
      |       ^~~~~~~~~~~~~~~
      |       rcl_timer_init
gmake[2]: *** [CMakeFiles/rclc_test.dir/build.make:191: CMakeFiles/rclc_test.dir/test/rclc/test_executor.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:188: CMakeFiles/rclc_test.dir/all] Error 2
gmake: *** [Makefile:149: all] Error 2
---
Failed   <<< rclc [1.22s, exited with code 2]

Summary: 0 packages finished [1.38s]
  1 package failed: rclc
  1 package had stderr output: 
``` 

#### Additional information

Changing the `rcl_timer_init2` to `rcl_timer_init` in both `~/rclc_ws/rclc/rclc/src/rclc/timer.c:36:18` and `~/rclc_ws/rclc/rclc/test/rclc/test_executor.cpp:575:7` did not resolve the issue.

Aditionally no info was found on this online hence why I'm opening up the issue.

**Micro-ROS**
When following the [Micro-ROS tutorial](https://micro.ros.org/docs/tutorials/core/first_application_linux/) there were no issues related to RCLC, but when trying to install just RCLC in a seperate workspace, this error occurs

Zard-C, 2023-09-19:
Hi, I can't reperduce this compile error :
```bash
rclc git:(humble) colcon build 
Starting >>> rclc    
Finished <<< rclc [30.1s]                       
Starting >>> rclc_lifecycle
Starting >>> rclc_parameter
Finished <<< rclc_lifecycle [15.5s]                                                                      
Finished <<< rclc_parameter [30.7s]                           
Starting >>> rclc_examples
Finished <<< rclc_examples [12.5s]                            

Summary: 4 packages finished [1min 14s]
```

But I believe your error `warning: implicit declaration of function ‘rcl_timer_init2’; did you mean ‘rcl_timer_init’? [-Wimplicit-function-declaration]` is caused by the lasted commit of `rclc: humble` version

Zard-C, 2023-09-19:
could you tell me the version of rcl? maybe rcl repo need to update.

NikolaZgonjanin, 2023-09-19:
How would I go about that? I can't find in the docs where it's located nor if there is a way to check the version. I'm presuming I have the latest one as I tested it on a freshly installed ROS2 Humble and it still gave me the error.

NikolaZgonjanin, 2023-09-19:
That was it, thank you very much. I forgot to switch branches when cloning again, so full mistakes on me 😅

closed 2023-09-19 by NikolaZgonjanin