ros2/rclc #394: [rclc/rclc_executor]: subscribe loaned message

Zard-C, 2023-10-12:
## Background

- OS: Ubuntu focal 
- Installation type: built from source 
- Version or commit hash: humble

When I was writting example codes that using loaned message with `rclc` interfaces (pub/sub) , the [publish_with_loaned_msg demo](https://github.com/Zard-C/rclc_example/blob/main/rclc_example/src/pub_loaned_msg.c) works fine, but then I found out that  there's no such implementation in rclc_executor which allow executor to take loaned subscription message from rmw. So I made a draft change to rclc_executor's implementation ([changes here](https://github.com/ros2/rclc/compare/humble...Zard-C:rclc:humble#diff-82aa34a48fc24ad5d3eefc4e7937b764b3ff91947b0e7394ffab7bc78def736b))

```diff
diff --git a/rclc/src/rclc/executor.c b/rclc/src/rclc/executor.c
index b91aa78..873790f 100644
--- a/rclc/src/rclc/executor.c
+++ b/rclc/src/rclc/executor.c
@@ -1104,20 +1104,39 @@ _rclc_take_new_data(rclc_executor_handle_t * handle, rcl_wait_set_t * wait_set)
     case RCLC_SUBSCRIPTION_WITH_CONTEXT:
       if (wait_set->subscriptions[handle->index]) {
         rmw_message_info_t messageInfo;
-        rc = rcl_take(
-          handle->subscription, handle->data, &messageInfo,
-          NULL);
-        if (rc != RCL_RET_OK) {
-          // rcl_take might return this error even with successfull rcl_wait
+        rcl_subscription_t* sub = handle->subscription;
+        if(rcl_subscription_can_loan_messages(sub)){
+          handle->data = NULL;
+          rc = rcl_take_loaned_message(sub, &handle->data, &messageInfo, NULL); // how to return it to rmw????
+          if (rc != RCL_RET_OK) {
+          // rcl_take_loaned_message might return this error even with successfull rcl_wait
           if (rc != RCL_RET_SUBSCRIPTION_TAKE_FAILED) {
-            PRINT_RCLC_ERROR(rclc_take_new_data, rcl_take);
+            PRINT_RCLC_ERROR(rcl_take_loaned_message, rcl_take_loaned_message);
             RCUTILS_LOG_ERROR_NAMED(ROS_PACKAGE_NAME, "Error number: %d", rc);
           }
-          // invalidate that data is available, because rcl_take failed
+          // invalidate that data is available, because rcl_take_loaned_message failed
           if (rc == RCL_RET_SUBSCRIPTION_TAKE_FAILED) {
             handle->data_available = false;
           }
           return rc;
+          }
+        }
+        else {
+          rc = rcl_take(
+            handle->subscription, handle->data, &messageInfo,
+            NULL);
+          if (rc != RCL_RET_OK) {
+            // rcl_take might return this error even with successfull rcl_wait
+            if (rc != RCL_RET_SUBSCRIPTION_TAKE_FAILED) {
+              PRINT_RCLC_ERROR(rclc_take_new_data, rcl_take);
+              RCUTILS_LOG_ERROR_NAMED(ROS_PACKAGE_NAME, "Error number: %d", rc);
+            }
+            // invalidate that data is available, because rcl_take failed
+            if (rc == RCL_RET_SUBSCRIPTION_TAKE_FAILED) {
+              handle->data_available = false;
+            }
+            return rc;
+          }
         }
       }
       break;
```
It worked with my example code [subscription_with_loaned_message](https://github.com/Zard-C/rclc_example/blob/main/rclc_example/src/sub_loaned_msg.c)

## Discussion 

1.  The draft used `rcl_take_loaned_message` to take loaned msg from rmw, the question is, when and where to  return it to rmw? I tried to [return  in sub callback](https://github.com/Zard-C/rclc_example/blob/d9fe4628cffa56586f507cefece3c36bda4a97d3/rclc_example/src/sub_loaned_msg.c#L14C21-L14C64), unfortunately, the return value was  `RCL_RET_ERROR`
2. The draft is only a attempt to use loaned message with rclc's subcription interface, and a lot of things need to be considered, do you have better ideas? CC  @JanStaschulat @fujitatomoya @pablogs9

Zard-C, 2023-10-13:
@JanStaschulat  Thanks Jan, before trying to use loaned message with rclc,  I took a look at rclcpp::executor's implementation([code here](https://github.com/ros2/rclcpp/blob/7f7ffcf6ed1143b4d4a9e0cf248e190731b2e5f8/rclcpp/src/rclcpp/executor.cpp#L638)), it shows where and when to take and return loaned message. I will try to figure out why `rcl_return_loaned_message_from_subscription` failed, it could be my mistake in my example codes.

By the way, my use-case of loaned message is:

Publish and subscribe a large but fixed(bounded) message in different nodes, a fixed picture generated by a camera, which may up to Mbytes.  I believed that using loaned message would reduce data copying.

fujitatomoya, 2023-10-13:
@Zard-C several things i would like to share with you for what you are trying to do here.

- `Loaned` interfaces are dependent on RMW implementation, so it might return `NOT SUPPORTED` when you call those interfaces. (if you are using `rmw_fastrtps_cpp`, it supports the feature.)
- besides, there is a condition. if you want to loan the message memory from RMW implementation buffer, message type must be PoD (Plane Old Data), that said message type have to be bound type (fixed size). so if we try to publish or subscribe Non-PoD or unbound message type such as `std::vector`, strings or arrays. it also returns error. because it has to allocate the dynamic memory buffer in RMW implementation memory pool always, and it does not know the size yet.

to see those conditions, we have interface to check if `can we use loaned interfaces for this publisher or subscription`, 

https://github.com/ros2/rcl/blob/45fc9520216013007c2d4f3bd69d778af9051aa3/rcl/include/rcl/publisher.h#L684-L691

https://github.com/ros2/rcl/blob/45fc9520216013007c2d4f3bd69d778af9051aa3/rcl/include/rcl/subscription.h#L869-L879

these are called in `rclcpp` before calling `rcl_take_loaned_message` to check the condition. and if it fails, it falls back to the default behavior which is RMW returns the copy of the data from own memry data.

be advised that using loaned interfaces are tricky, this is all because of the memory ownership.

https://github.com/ros2/rcl/blob/45fc9520216013007c2d4f3bd69d778af9051aa3/rcl/include/rcl/subscription.h#L643-L644

i want to point out above constraint. this says, once the message is returned via `rcl_return_loaned_message`, we cannot keep the data reference since the physical memory is owned by the RMW implementation. if the application keeps the pointer somewhere in the application for later use, the data would not be valid because RMW implementation needs to reuse the memory pool it owns. so if the application is using the loaned interface, the above constraints must be applied. instead, data corruption can be seen in the application.

further more, loaned interfaces are the memory management between ROS 2 and RMW implementation endpoints. it does not say it achieve the zero copy between the endpoints underneath implementation. if your final target is to support zero copy data sharing, you need to enable RMW implementation data sharing feature. you can see more details for https://github.com/ros2/rmw_fastrtps#enable-zero-copy-data-sharing.

i hope this helps some, if you have any questions, let me know. I am positive for this feature enabled in `rclc` as well.