ros2/rclc #401: [Rolling] Build error rclc packages

JanStaschulat, 2024-03-08:
The build job on open robotics build farm has been failing since Wednesday (06.03.2024). It has failed last three days.

https://build.ros2.org/job/Rsrc_uJ__rclc__ubuntu_jammy__source/7/

Error message: ` Remote branch debian/ros-rolling-rclc_6.1.0-2_jammy not found in upstream origin`

Maybe some issue on build farm?

```
23:49:35 # BEGIN SUBSECTION: get sources
23:49:37 Invoking 'git clone --branch debian/ros-rolling-rclc_6.1.0-2_jammy --depth 1 --no-single-branch https://github.com/ros2-gbp/rclc-release.git /tmp/sourcedeb/source'
23:49:37 Cloning into '/tmp/sourcedeb/source'...
23:49:37 fatal: Remote branch debian/ros-rolling-rclc_6.1.0-2_jammy not found in upstream origin
23:49:37 # END SUBSECTION
```
https://build.ros2.org/job/Rsrc_uJ__rclc__ubuntu_jammy__source/13/console

Zard-C, 2024-03-08:
```shell
subprocess.CalledProcessError: Command '['git', 'clone', '--branch', 'debian/ros-rolling-rclc_6.1.0-2_jammy', '--depth', '1', '--no-single-branch', 'https://github.com/ros2-gbp/rclc-release.git', '/tmp/sourcedeb/source']' returned non-zero exit status 128.
```

JanStaschulat, 2024-03-08:
Thanks for debugging, @Zard-C . Anything I can do? Or should I contact someone who is responsible for the build farm?

christophebedard, 2024-03-08:
Rolling/Jazzy is moving from Ubunty Jammy to Ubuntu Noble: https://discourse.ros.org/t/preparing-ros-2-rolling-for-the-transition-to-ubuntu-24-04/35673. I think the move is not 100% done, so failures like that are expected.

closed 2024-10-31 by JanStaschulat