ros2/rclc #405: Unclear how to use use_sim_time

PatrickHaecker, 2024-07-01:
I haven't found in the documentation how to activate [`use_sim_time`](http://design.ros2.org/articles/clock_and_time.html#ros-time) in `rclc`.
Any hints are appreciated. But in the end this should probably be a paragraph in the documentation or, depending on complexity, shown with an example.

fujitatomoya, 2024-07-02:
i believe that currently `rclc` only supports the clock type with `RCL_STEADY_TIME`. probably we need to design the clock support for `RCL_SYSTEM_TIME` and `RCL_ROS_TIME`.

PatrickHaecker, 2024-07-03:
Based on reading the source I agree that `rclc` probably does not support `RCL_ROS_TIME`. However, I also learned, that I do not need it for my use case.

For everyone with a similar problem, I recommend the following:

1. If you msg files contain the `ROS time`, you might not need `use_sim_time` support. Remember, that time values can be in `std_msgs/Header` in a or multiple field(s) somewhere in your msg definition or somewhere deeply nested within complex fields.
2. If you need the `ROS time`, chances are that you can just subscribe to `/clock` using type `rosgraph_msgs/msg/Clock`. `ros2 bag play --clock`  uses a regular publisher to publish that message periodically (depending on a parameter when calling the binary and the play speed) and it can be read by a regular subscription. To transfer the value between the callbacks, you can save the value in a global variable, a pointer, a shared reference,  a closure or whatever your programming language offers.
3. If you want to kind of hide that subscriber, you can wrap it in a callback and call your real regular callback with the time as an additional argument.
4. The implementation of most of the ROS time is not done in rcl, but in the language-specific client libraries like `rclcpp` and `rclpy`. If you want to go down deep into the rabbit hole, it's implementing the subscription to `/clock` in a separate thread (possibly using `pthreads`) (otherwise `ROS time` never gets updated in a long running callback making `ROS time` only useful directly after the start of the callback which means the above solution should do) and updating all clocks which use `ROS time` (via a list of all such clocks like in an event listener and using synchronization mechanisms due to multithreading). The corresponding `rclcpp` code can be found in [`rclcpp time_source.cpp`](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/time_source.cpp).

closed 2024-07-03 by PatrickHaecker